veripool/verilog-mode #1664: Order of ports in instance -> how to reorder?

ldm1417, 2020-03-20:
Hi All,

Is there a way to reorder the ports list in the instance? Could it be done through the TEMPLATE of AUTOINST?

Thank you!

wsnyder, 2020-03-20:
Perhaps this does what you are looking for? https://www.veripool.org/projects/verilog-mode/wiki/Verilog-mode-Help#verilog-auto-inst-sort

ldm1417, 2020-03-20:
When I set the verilog-auto-inst-sort to true (see below), the ports in the instance are sorted alphabetically. 
**verilog-auto-inst-sort:t**
When I doesn't use this variable at all (not declared), the ports in the instance are sorted in the same manner as in the referenced module, but they are divided to Inputs and outputs.
But, I'd like to have the same ports order in the instance as in the module, without additional division of them to Inputs and Outputs. Is this possible?
Thanks!

wsnyder, 2020-03-20:
There is no way to do this presently.  Note Verilog-Mode actually needs the input/output comments so any reordering would still need them.

fastwind0, 2020-07-22:
> But, I'd like to have the same ports order in the instance as in the module, without additional division of them to Inputs and Outputs. Is this possible?

I met the same issue. But I don't know how to use "verilog-auto-inst-sort:t". Need I add it to the bottom of file? or use it in the command line?

wsnyder, 2020-07-28:
You probably figured this out already, but you put it in the bottom of the Verilog file:

```
// Local Variables:
// verilog-auto-inst-sort: t
// End:
```

closed 2020-07-28 by wsnyder