veripool/verilog-mode #1666: verilogmode re-indenting certain keywords

rasmus-madsen, 2020-04-08:
hi -
even with the newest verilog-mode.el

many keywords are re-indented when going to new line with "enter"
some don't

example 
```
module my_module begin

endmodule
```

if I set the curser right after the begin keyword and hit "enter"
it will indent this line to 

```
   module my_module begin

endmodule
```

it it will keep doing this as many times as do this
same goes for class and if begin etc.

it annoys me alot - and I seem to remember it didn't used to do that.
it should always align to the block it is in.

could anyone take a look or guide me to where I can look to solve this issue.

wsnyder, 2020-04-08:
This isn't legal syntax, so how it will get indented is basically random.

rasmus-madsen, 2020-04-08:
you are right..  I was mixing two different examples.
but the problem is still very much existing.

I have updated the example with proper syntax

wsnyder, 2020-04-08:
Please send the output from M-x  verilog-submit-bug-report RETURN

rasmus-madsen, 2020-04-08:
@wsnyder as promised
I tried to turn a lot of the indention off hoping to solve the issue. but with no luck.
it might be a conflicting option but I have not been able to find it.
and resetting everything does not seem to help.
(btw thank you for the great work on verilog mode you and I met last year in munic at the chips alliance conference)
== The code: ==

Emacs  : GNU Emacs 25.3.1 (x86_64-pc-linux-gnu, GTK+ Version 2.24.23)
 of 2018-05-15
Package: verilog-mode v2020-03-24-0d62192-vpo

current state:
==============
(setq
 verilog-active-low-regexp nil
 verilog-after-save-font-hook nil
 verilog-align-ifelse nil
 verilog-assignment-delay ""
 verilog-auto-arg-sort nil
 verilog-auto-declare-nettype nil
 verilog-auto-delete-trailing-whitespace nil
 verilog-auto-endcomments t
 verilog-auto-hook nil
 verilog-auto-ignore-concat nil
 verilog-auto-indent-on-newline t
 verilog-auto-inout-ignore-regexp nil
 verilog-auto-input-ignore-regexp nil
 verilog-auto-inst-column 40
 verilog-auto-inst-dot-name nil
 verilog-auto-inst-interfaced-ports nil
 verilog-auto-inst-param-value nil
 verilog-auto-inst-sort nil
 verilog-auto-inst-template-numbers nil
 verilog-auto-inst-vector t
 verilog-auto-lineup 'declarations
 verilog-auto-newline 0
 verilog-auto-output-ignore-regexp nil
 verilog-auto-read-includes nil
 verilog-auto-reset-blocking-in-non t
 verilog-auto-reset-widths t
 verilog-auto-save-policy nil
 verilog-auto-sense-defines-constant nil
 verilog-auto-sense-include-inputs nil
 verilog-auto-star-expand t
 verilog-auto-star-save nil
 verilog-auto-template-warn-unused nil
 verilog-auto-tieoff-declaration "wire"
 verilog-auto-tieoff-ignore-regexp nil
 verilog-auto-unused-ignore-regexp nil
 verilog-auto-wire-type nil
 verilog-before-auto-hook nil
 verilog-before-delete-auto-hook nil
 verilog-before-getopt-flags-hook nil
 verilog-before-save-font-hook nil
 verilog-cache-enabled t
 verilog-case-fold t
 verilog-case-indent 0
 verilog-cexp-indent 2
 verilog-compiler "echo 'No verilog-compiler set, see \"M-x describe-variable verilog-compiler\"'"
 verilog-coverage "echo 'No verilog-coverage set, see \"M-x describe-variable verilog-coverage\"'"
 verilog-delete-auto-hook nil
 verilog-getopt-flags-hook nil
 verilog-highlight-grouping-keywords nil
 verilog-highlight-includes t
 verilog-highlight-modules nil
 verilog-highlight-translate-off nil
 verilog-indent-begin-after-if 0
 verilog-indent-declaration-macros nil
 verilog-indent-level 2
 verilog-indent-level-behavioral 0
 verilog-indent-level-declaration 0
 verilog-indent-level-directive 0
 verilog-indent-level-module 0
 verilog-indent-lists t
 verilog-library-directories '(".")
 verilog-library-extensions '(".v" ".va" ".sv")
 verilog-library-files nil
 verilog-library-flags '("")
 verilog-linter "my_lint_shell_command"
 verilog-minimum-comment-distance 40
 verilog-mode-hook '((lambda nil (font-lock-mode 1)))
 verilog-mode-release-emacs nil
 verilog-mode-version "2020-03-24-0d62192-vpo"
 verilog-preprocessor "verilator -E __FLAGS__ __FILE__"
 verilog-simulator "echo 'No verilog-simulator set, see \"M-x describe-variable verilog-simulator\"'"
 verilog-tab-always-indent t
 verilog-tab-to-comment nil
 verilog-typedef-regexp nil
 verilog-warn-fatal nil
 )

wsnyder, 2020-04-08:
I don't see anything obvious going wrong when I use those settings, but maybe I'm trying the wrong thing.  With the below (or feel free to edit) where is your cursor and what do you type and what looks wrong?

```
module my_module #(
) (
);
endmodule
```

rasmus-madsen, 2020-04-08:
I create a file called tmp.sv
I copy the code above.
I place my curser after #( and press enter to create a new line.

here is my code after that
```
                    module my_module #(
                                       
                                                         
                                      
) (
);
endmodule
```
and here it is after repeating the step

```
                                       module my_module #(
                                                          
                                       
                                                         
                                      
) (
);
endmodule
```

wsnyder, 2020-04-08:
I don't see that behavior.  Does it say "verilog-mode" at the bottom?  Try starting with

emacs -q --no-site-file my.sv

and see if it works.  If it does, remove the no-site-file, then if that works remove the -q.  You may also need to load the most recent verilog mode (-l path/to/verilog-mode.el)

rasmus-madsen, 2020-04-09:
it does say verilog mode.
I finally figured out what is causing the trouble.

I realize this is out of the scope for this site. 
but if there is any chance you can point me in a direction for a solution it would be much appreciated.

the issue was I have some custom keymaps in verilog mode for easy insertion of uvm 

```
(define-prefix-command 'verilog-mode-map)
(global-set-key (kbd "C-p") 'verilog-mode-map)

;; to use any of these press ctrl+p - m/h/l/e/w/f
(define-key verilog-mode-map (kbd "C-p h") "`uvm_info(tID, $sformatf(\"\"), UVM_HIGH)")
(define-key verilog-mode-map (kbd "C-p m") "`uvm_info(tID, $sformatf(\"\"), UVM_MEDIUM)")
(define-key verilog-mode-map (kbd "C-p l") "`uvm_info(`gfn, $sformatf(\"\"), UVM_LOW)")
(define-key verilog-mode-map (kbd "C-p e") "`uvm_error(tID, $sformatf(\"\"))")
(define-key verilog-mode-map (kbd "C-p f") "`uvm_fatal(tID, $sformatf(\"\"))")
```

defining the prefix causes the issue.
`(define-prefix-command 'verilog-mode-map)`

I can't really figure out why this would cause the indentation schemes to go crazy?

wsnyder, 2020-04-09:
Even with that I still don't see  a problem, but I can imagine it might cause a problem as the map is already installed as the root keymap.  Presumably what you really wanted was to make and use my-verilog-mode-map so it doesn't conflict with the old one.

closed 2020-04-09 by wsnyder

rasmus-madsen, 2020-04-09:
thank you tried other map names with no change.. I will debug this on my own