veripool/verilog-mode #1667: Replacing instance port parameters with it's values in parent module ports - An issue with usage of verilog-auto-inst-param-value

engrvns, 2020-04-21:
Hi Wilson,
I am finding an issue with the verilog-mode where the parent module port definitions inherit the child parameter names. 

This is inefficient and forces the user to -
1) redefine the instance parameters at the instance level which is redundant
2)Difficult to keep updating as ExampInst could be owned by another owner
3)Difficult to maintain if there are lots of parameters and multiple instances...

In the example below, the ports of ExampInst are defined in terms of W.
Which would mean ExampInst wont's compile unless parameter W is re-defined.

This behavior happens due to requirements of verilog-auto-inst-param-value:t
How can we avoid this issue?

Example:
module InstModule 
  #(parameter W = 2)
  (
   input [W-1:0] i, 
   output[W-1:0] o 
   );
endmodule

module ExampInst 
  (
   /*AUTOINPUT*/
   // Beginning of automatic inputs (from unused autoinst inputs)
   input [W-1:0]	i,			// To instName of InstModule.v
   // End of automatics
   /*AUTOOUTPUT*/
   // Beginning of automatic outputs (from unused autoinst outputs)
   output [W-1:0]	o			// From instName of InstModule.v
   // End of automatics
   );
  InstModule
    instName (/*AUTOINST*/
	      // Outputs
	      .o			(o[W-1:0]),
	      // Inputs
	      .i			(i[W-1:0]));
endmodule // ExampInst

// Local variables:
// verilog-auto-sense-defines-constant:t
// verilog-auto-inst-param-value:t
// End:

wsnyder, 2020-04-21:
What behavior are you suggesting/recommending?

MoSaifuddin, 2024-08-12:
I agree with the issue described. I am facing the same issue.

@wsnyder ,
When verilog-auto-inst-param-value is non-nil and no parameter is set at the instantiation level, verilog-auto should pick up the default parameter value. Here is what I'd expect:

`module InstModule
#(parameter W = 2)
(
input [W-1:0] i,
output[W-1:0] o
);
endmodule

module ExampInst
(
/AUTOINPUT/
// Beginning of automatic inputs (from unused autoinst inputs)
input [W-1:0] i, // To instName of InstModule.v
// End of automatics
/AUTOOUTPUT/
// Beginning of automatic outputs (from unused autoinst outputs)
output [W-1:0] o // From instName of InstModule.v
// End of automatics
);
InstModule
instName (/AUTOINST/
// Outputs
**.o (o[1:0]),**
// Inputs
**.i (i[1:0]));**
endmodule // ExampInst

// Local variables:
// verilog-auto-sense-defines-constant:t
// verilog-auto-inst-param-value:t
// End:`