veripool/verilog-mode #1669: How to write a AUTO rule for creating AUTOOUTPUTS of internal signals

engrvns, 2020-05-03:
AUTOOUTPUTS are normally inferred out of "non-internally" connected instance outputs.
Is there any way one could write a AUTO to force creation of output ports based of internal signal names.

One usage for this for this when you want to width-cast an instance output before it gets connected to a module output. This forces the user to create an internal signal.

Simplistic Example:
module top (output [3:0] out ...);

  /*out AUTO_TEMPLATE (
    .out (temp_out[]),
  );*/

  assign out = `4(temp_out); //Width casting forces internal signal
  
   inst inst ( 
      .out(temp_out[1:0))
      ...
     );
endmodule

engrvns, 2020-05-03:
The answer is verilog-auto-output-every

wsnyder, 2020-05-04:
Glad you found the answer in AUTOOUTPUTEVERY.  I assume that worked for your use case.

closed 2020-05-04 by wsnyder

engrvns, 2020-05-07:
Spoke too soon. AUTOOUTPUTEVERY seems to work only for AUTOWIRES.
the above example does not have AUTOWIRE and does not seem to work

How do we force the signal "out" in the example to appear as a output port.

wsnyder, 2020-05-08:
Yes, you need AUTOWIRE, in theory adding it should not hurt anything.