veripool/verilog-mode #1673: Use of Ordered AUTO_TEMPLATEs to simplify code integration

engrvns, 2020-06-06:
This is an enhancement proposal for existing verilog-mode which will be useful when code has Multiple instances of a Module(s) with large number of ports and deviations to a non-trivial subset of ports.

The current method(s) to handle this is -

Method 1: Define a common denominator AUTO_TEMPLATE for the module and define override custom connections at instantiation.
This definitely works. But custom connections would means the advantage of rule based connections are lost...

Method 2: Duplicate the entire AUTO_TEMPLATE and modify the rules for the subset,
This works as well. But code duplication, bloat and maintenance.

Am method would be to use ordered AUTO_TEMPLATE and have precedence rules define the final AUTO_TEMPLATE
1) Define an common denominator AUTO_TEMPLATE
2) Define an overlay AUTO_TEMPLATE for the "exception subset" and have precedence.

Refer attached example for a proposal ...

[OverlayExample.txt](https://github.com/veripool/verilog-mode/files/4740732/OverlayExample.txt)

wsnyder, 2020-06-06:
I can see how that's useful.  I'd suggest instead that the "overlays" have some method to specify the "base". This is more flexible and easier to implement.  The base is just then a normal AUTO_TEMPLATE.  Probably want to use the word "extends" somehow as that's what SystemVerilog classes use.

```
/* base AUTO_TEMPLATE \(.*\) ( <- base auto_template
   .\(portA\) (\1_ruleA),
   .\(portB\) (\1_ruleB),
   .\(portC\) (\1_ruleC),
);*/  
/* mod1 AUTO_TEMPLATE \(.*\) AUTO_EXTENDS(base) (
  .\(portB\) (\1_ruleB0), <- Rule based deviation for portB 
);*/
```

engrvns, 2020-06-06:
Wilson,

Thanks for the response

Class extension. Good idea.
I assume that the same extends to AUTOINSTPARAMS as it's in the AUTO_TEMPLATE section...
It's not possible in the current verilog-mode?

IMHO, It will have it's limitations, which would mean that the base template needs to get redefined every time there is an exception.
But that should be OK...

Engr

engrvns, 2020-06-29:
Hi Wilson,

When do you think this feature will get into the release.
I can test it out if there is a beta version.

- engrvns