veripool/verilog-mode #1682: Sub module has ifdef for IO

Mrainy, 2020-06-18:
In my sub module, I have

module sub (
`ifdef AAA
input AAA,
`endif
`ifdef BBB
input BBB,
`endif
);
endmodule

When I use AUTOINST to instantiate this module, is there a way to use verilog-mode to get tonly he input AAA from this submodule?

wsnyder, 2020-06-18:
verilog-mode ignores ifdefs (see the FAQ) so will see both. You will need to ifdef where ypu use it, or use a regexp to select AAA as described in the verilog-auto-inst help.

Mrainy, 2020-06-20:
Thanks for the quick reply. Would you help to provide an example how to select AAA ONLY? Basically, if I use AUTOINST, the signal BBB is always propagated during instantiation. Probably I can change the question to: how to ignore some module IO with regexp with AUTOINST?

wsnyder, 2020-06-20:
Oh, sorry, only auto-inst-param takes a regexp, I'll fix that.

wsnyder, 2020-06-27:
You can now use e.g.

```
   sub  sub1
       (/*AUTOINST("ia")*/
```

closed 2020-06-27 by wsnyder

Mrainy, 2020-07-02:
Is this feature to list the IO we want? How about the ones we want to remove? Something like 
`sub  sub1
       (/*AUTOINST(not_include("ia"))*/`

wsnyder, 2020-07-05:
I updated it to allow "?!" in front of a regexp to now invert it, that is use

```
/* sub  sub1
       (/*AUTOINST("?!ia")*/
);
*/
```

Your template idea is also a good one but more work than I'd like at the moment.

zjuxieyi, 2020-08-14:
Hi, wsnyder 

Is there a quick way to append more ports in the regexp?
Because, actually I have hundreds of ports which is present/unpresent controlled by hundreds of macro-definitions.

Thanks,
Alex

wsnyder, 2020-08-14:
If you have a lot of exclusions hopefully you can instead ifdef them out.

```
/* sub  sub1
       (/*AUTOINST("?!ia")*/
`ifdef NEVER
        .ignore1,
        .ignore2,
         ...
`endif
);
*/
```

zjuxieyi, 2020-08-14:
> If you have a lot of exclusions hopefully you can instead ifdef them out.
> 
> ```
> /* sub  sub1
>        (/*AUTOINST("?!ia")*/
> `ifdef NEVER
>         .ignore1,
>         .ignore2,
>          ...
> `endif
> );
> */
> ```

Thanks for your help, wsnyder .

Do you mean explicitly write the exclusion when instance it? Or add that in AUTO_TEMPLATE?

sub  sub1 (

`ifdef HAS_THESE_PINS
        .ignore_output (ignore_output), //do the right connection, eg, pull them to my top
        .ignore_input  (ignore_input), //do the right connection, eg, pull them to my top
`else
        .ignore_output (), //leave them as dangle
        .ignore_input  (32'b0), //tieoff them
         ...
`endif
       /*AUTOINST*/
);

wsnyder, 2020-08-14:
It needs to be below the AUTOINST, not in the template.

zjuxieyi, 2020-08-15:
> It needs to be below the AUTOINST, not in the template.

Hi, wsnyder 
Sorry, I'm not quite clear about this.

I tried to add something like this:

sub  sub1
       (/*AUTOINST("?!ia")*/
);

It doesn't work.

Thanks,
Alex

wsnyder, 2020-08-15:
That should work, there's passing tests that check it.  Make sure M-x describe-variable verilog-mode-version RETURN  shows 2020-07-05 or newer?