veripool/verilog-mode #1685: wront indentation on constraints

luisgutz, 2020-07-16:
The closing brace of an constraint block is not recognize as finishing the indentation, and they keep piling up.

This is how it looks:
```
class alarmCfgC;
  rand bit [15:0] v1;
  rand bit [15:0] v2;
  rand bit [1:0] c1;

  constraint c1 { (c1 == 'b00  ) -> (v1  < v2);
    (c1 == 'b01 ) -> (v1  > v2);
    (c1 == 'b10) -> (v1 == v2);}
    constraint c1 { (c1 == 'b00  ) -> (v1  < v2);
      (c1 == 'b01 ) -> (v1  > v2);
      (c1 == 'b10) -> (v1 == v2);}
      
      endclass // alarmCfgC
```

and this is what I was expecting:
```
class alarmCfgC;
  rand bit [15:0] v1;
  rand bit [15:0] v2;
  rand bit [1:0] c1;

  constraint c1 { (c1 == 'b00  ) -> (v1  < v2);
    (c1 == 'b01 ) -> (v1  > v2);
    (c1 == 'b10) -> (v1 == v2);}
constraint c1 { (c1 == 'b00  ) -> (v1  < v2);
  (c1 == 'b01 ) -> (v1  > v2);
  (c1 == 'b10) -> (v1 == v2);}
  
endclass // alarmCfgC
```

I can see that '{' is the first element in verilg-indent-re const, but I cannot find where the end of indentation is calculated

acr4, 2020-07-16:
Indention works for me with both versions `2020-02-23-dddb795-vpo-GNU` and `2020-06-27-0da9923-vpo-GNU`.  What version are you using (`C-h v verilog-mode-version RET`)?

luisgutz, 2020-07-16:
I'm on 2002.02.23 (436).
I'll update to the latest.

luisgutz, 2020-07-16:
so Emacs was loading a far older version.
When using the latest from github, the constraints are correctly indented.

I would close this ticket, but I don't think I can.

closed 2020-07-16 by wsnyder