veripool/verilog-mode #1686: vexpand using Generate statement + Multidimensional array

meka2405, 2020-08-06:
Hello,

 I am trying to instance module2 below using vexpand. Not sure how to use vexpand in case of generate statement + multidimensional array

module top();

parameter NUM_OF_INST = 5;

wire [NUM_OF_INST -1:0][63:0] mult_port_signal;
wire one_bit_signal;

module1 mod1_i (
.one_bit_signal (one_bit_signal),   // Input to this module
.mult_port_signal (mult_bit_signal) //output from this module
);

genvar gi;
generate 
  for (gi=0; gi<NUM_OF_INST; gi=gi+1) begin : NI
    module2 module2_i (
       .one_bit_signal (one_bit_signal),   // but only when gi=0, else don't connect to anything
       .per_inst_signal (mult_bit_signal[$gi][63:0])  // one bit per instance.
    );
end
endgenerate

endmodule

wsnyder, 2020-08-06:
"gi" only changes at elaboration time, so you need to write the code to be different based on gi using normal Verilog conventions, so use a if statement in the loop.

```
generate 
  for (gi=0; gi<NUM_OF_INST; gi=gi+1) begin : NI
    if (gi == 0) begin
/*
module2 AUTO_TEMPLATE (
???????
);
*/
      module2 module2_i (/*AUTOINST*/
         .one_bit_signal (one_bit_signal),   // but only when gi=0, else don't connect to anything
         .per_inst_signal (mult_bit_signal[gi][63:0])  // one bit per instance.
      );
    end
    else begin
/*
module2 AUTO_TEMPLATE (
???????
);
*/
      module2 module2_i (/*AUTOINST*/
         .one_bit_signal (one_bit_signal),   // but only when gi=0, else don't connect to anything
         .per_inst_signal (mult_bit_signal[gi][63:0])  // one bit per instance.
      );
    end
end
endgenerate
```

closed 2020-08-06 by wsnyder

meka2405, 2020-08-06:
Thanks for the prompt response. One followup question. I am assuming AUTOWIRE for mult_bit_signal doesn't work either. correct? When I tried, its getting defined as below.

wire [gi][63:0] mult_bit_signal;

instead of 

wire[NUM_OF_INST-1:0][63:0] mult_bit_signal;

wsnyder, 2020-08-06:
Verilog-mode doesn't parse generate loops - generally AUTOWIRE is only useful when the wires exactly match what is instantiated and in many cases you need to manually declare them instead.