veripool/verilog-mode #1693: AUTOINOUTPARAM does not include parameter assignment

scdemarc, 2020-08-24:
The use of 
/\*AUTOINOUTPARAM("dut_module")*/
populates
parameter PARAM_NAME;

instead of
parameter PARAM_NAME=VALUE;

The former causes Cadence xcelium to fail with a compile error.
It appears that value assignment is required in order for the automated
parameter insert to be usable.

wsnyder, 2020-08-24:
This is noted in the documentation for it:

  Parameters do not have values, which is SystemVerilog 2009 syntax.

I appreciate this might be annoying, but Verilog-mode doesn't know how to set the defaults so it thus requires you then specify all parameters on the level above.  If this isn't what you want you'll have to manually specify the parameters.  Unfortunately Verilog doesn't have a "parameter FOO = default" which would take the value from the level below.

closed 2020-08-24 by wsnyder

scdemarc, 2020-08-24:
A dummy assignment apart from the real value
from the reference module could be an effective
workaround to having to add the params manually.
The real value would be propagated at compile-time,
(Eg- as in the case of a bind directive)
while the dummy assignment to a fictitious value 
would simply allowing for avoiding the compile error
while retainIng the use of the auto-insert feature.

acr4, 2020-08-25:
I tend to agree with Wilson on this one -- valueless parameter declarations are legal during compilation, and have been for over a decade.  The burden here falls on Cadence to fix their tool.

bcrules82, 2022-11-29:
Would it be possible to add a variable to optionally grab the default assignments (if they exist) as-is from the module? I'm working with a highly parameterized verification testbench that mirrors the DUT's parameter list, but unfortunately the UVM macro infrastructure doesn't work if parameterized classes lack a default specialization.

wsnyder, 2022-11-29:
Verilog-mode doesn't currently parse the parameter values, and this isn't something likely to be added, sorry.