veripool/verilog-mode #1695: use AUTO_TEMPLATE for multiple times

zjuxieyi, 2020-09-16:
Hi,
I have a sub module, which will be instantiated for several times, but its instance-name and top-connection has a few difference.
I'd like to use AUTO_TEMPLATE , because actually there are hundreds of pins for me to connect.
How to handle this case?

/*sub_module AUTO_TEMPLATE (
  .in(in),
  .out(out),
  .common(com)
); */

sub_module u_sub1 (
  .in (host1_in),
  .out(host1_out),
  .common(com),
  ..
 );

sub_module u_sub2 (
  .in (host2_in),
  .out(host2_out),
  .common(com),
  ..
 );

Thanks,
Alex

wsnyder, 2020-09-16:
Doesn't the @ syntax do what you want - please see the FAQ and help pages if you don't know what I mean.

zjuxieyi, 2020-09-16:
Yes, @ stands for a number, [0-9]+
For these 2 instances, not only their instance-name are different, and simply add a postfix,  but also their pin's top connections are different. 

For a single submodule, is it OK to write its dedicated AUTO_TEMPLATE twice? or even multiple times?

/*sub_module AUTO_TEMPLATE (
.in(rule1_for_in),
.out(rule1_for_out),
.common(com)
); */
sub_module u_sub1 (
/*AUTOINST*/
);

//write the same module twice for the 2nd instance
/*sub_module AUTO_TEMPLATE (
.in(rule2_for_in),
.out(rule2_for_out),
.common(com)
); */
sub_module u_sub2 (
/*AUTOINST*/
);

wsnyder, 2020-09-16:
Yes.

closed 2020-09-16 by wsnyder

zjuxieyi, 2020-09-17:
Hi, wsnyder 

Do u mean, for a single module, its corresponding AUTO_TEMPLATE could be write more than once?
And the latest one's TEMPLATE will overwrite the previous one, and it only impacts its subsequent AUTOINST?
Is my understanding right?

Thanks,
Alex

wsnyder, 2020-09-17:
Correct, please see the documentation on this: [verilog-auto-inst](https://www.veripool.org/projects/verilog-mode/wiki/Verilog-mode-Help#verilog-auto-inst) "When an instantiation is expanded verilog-mode simply searches up for the closest template.