veripool/verilog-mode #1698: Sensing Multidimensional Port Containing Multiplication Operator

alanamckee, 2020-10-23:
Hi, from the documentation, I see that multidimensional I/O has only experimental support, and I'm unsure if this is a known issue.

What I've encountered is if one of the dimensions in a multidimensional input/output contains the multiplication operator, it is not brought up the hierarchy with /*AUTOINPUT*/ or /*AUTOOUTPUT*/:

The initiation comment provided by /*AUTOINST*/ is correct and all the ports are present. Using the add, subtract, and divide operators causes no problems with /*AUTOINPUT*/ and /*AUTOOUTPUT*/.

I am using verilog-mode version 2020-09-22-03ac87a-vpo.

```
module sub_mod
(
   input [8+4-1:0][7:0] add_left,
   input [8-4-1:0][7:0] substract_left,
   input [8*4-1:0][7:0] multiply_left,
   input [8/4-1:0][7:0] divide_left

   input [7:0][8+4-1:0] add_right,
   input [7:0][8-4-1:0] substract_right,
   input [7:0][8*4-1:0] multiply_right,
   input [7:0][8/4-1:0] divide_right,
);
endmodule : sub_mod

module top_mod
(
   /*AUTOINPUT*/
   // Beginning of automatic inputs (from unused autoinst inputs)
   input [8+4-1:0] [7:0]      add_left,           // To sub_mod_i of sub_mod.v
   input [7:0] [11:0]         add_right,        // To sub_mod_i of sub_mod.v
   input [8/4-1:0] [7:0]      divide_left,        // To sub_mod_i of sub_mod.v
   input [7:0] [1:0]          divide_right,     // To sub_mod_i of sub_mod.v
   input [8-4-1:0] [7:0]      substract_left,   // To sub_mod_i of sub_mod.v
   input [7:0] [3:0]          substract_right // To sub_mod_i of sub_mod.v
   // End of automatics
);

   sub_mod sub_mod_i
      (/*AUTOINST*/
       // Inputs
       .add_left                        (add_left/*[8+4-1:0][7:0]*/),
       .substract_left			(substract_left/*[8-4-1:0][7:0]*/),
       .multiply_left			(multiply_left/*[8*4-1:0][7:0]*/),
       .divide_left			(divide_left/*[8/4-1:0][7:0]*/),
       .add_right			(add_right/*[7:0][8+4-1:0]*/),
       .substract_right			(substract_right/*[7:0][8-4-1:0]*/),
       .multiply_right			(multiply_right/*[7:0][8*4-1:0]*/),
       .divide_right			(divide_right/*[7:0][8/4-1:0]*/));

endmodule : top_mod
```

wsnyder, 2020-10-23:
Bug was a side effect of looking for the */.  Fixed, I think, thanks for the good test case.

closed 2020-10-23 by wsnyder

alanamckee, 2020-10-28:
This fix worked. Thank you.