veripool/verilog-mode #1700: Fails to build on CentOS 8 w/ emacs 26.1 and XEmacs 21.5

iperryman-xeda, 2020-11-04:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (You may want to attach output from 'M-x verilog-submit-bug-report')

Can you please check your github name is set to your real name (click on your avatar icon in upper right, then "settings" then "Name")?

Not sure if your Travis CI includes CentOS 8 ....  I could not find a spec that indicates what systems you are building against for the Travis "passed" flag on the main page.

I think this is all the relevant info you need to repro this issue:

Here is the error first ... below is the info about the system and the verilog-mode git repo.

```
13:58:14 [iperryman@desk-03 verilog-mode]$ make
rm -rf e
mkdir e
./config_rev.pl . . <verilog-mode.el >e/verilog-mode.el
emacs -batch -q -f batch-byte-compile e/verilog-mode.el
Loading /usr/share/emacs/site-lisp/site-start.d/autoconf-init.el (source)...
Loading /usr/share/emacs/site-lisp/site-start.d/desktop-entry-mode-init.el (source)...
rm -rf x
mkdir x
./config_rev.pl . . <verilog-mode.el >x/verilog-mode.el
xemacs -batch -q -f batch-byte-compile x/verilog-mode.el

  Loading leim-list...
Compiling /home2/iperryman/elisp/verilog-mode/x/verilog-mode.el...
Loading skeleton...
While compiling verilog-define-abbrev in file /home2/iperryman/elisp/verilog-mode/x/verilog-mode.el:
  ** define-abbrev called with 6 arguments, but accepts only 2-5
While compiling verilog-set-compile-command:
  ** #'format called with a non-constant CONTROL-STRING, (eval verilog-tool)
While compiling toplevel forms:
  !! Wrong type argument ((sequencep verilog))
>>Error occurred processing x/verilog-mode.el: Wrong type argument: sequencep, verilog

Done
make: *** [Makefile:124: x/verilog-mode.elc] Error 1
[2]+  Done                    xemacs

```

### 
```
14:00:17 [iperryman@desk-03 verilog-mode]$ emacs --version
GNU Emacs 26.1
Copyright (C) 2018 Free Software Foundation, Inc.
GNU Emacs comes with ABSOLUTELY NO WARRANTY.
You may redistribute copies of GNU Emacs
under the terms of the GNU General Public License.
For more information about these matters, see the file named COPYING.

14:07:50 [iperryman@desk-03 verilog-mode]$ xemacs --version

  Loading leim-list...
XEmacs 21.5  (beta34) "kale" c0ed7ef9a5a1 [Lucid] (x86_64-redhat-linux, Mule) of Thu Jan  9 2020 on buildvm-23.phx2.fedoraproject.org

14:10:48 [iperryman@desk-03 verilog-mode]$ lsb_release -a**
LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	CentOS
Description:	CentOS Linux release 8.2.2004 (Core) 
Release:	8.2.2004
Codename:	Core

14:11:08 [iperryman@desk-03 verilog-mode]$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
14:16:57 [iperryman@desk-03 verilog-mode]$ git log
commit a2a685e19f0c863b4d07364b9ac3bd2be9552273 (HEAD -> master, origin/master, origin/HEAD)
Author: Wilson Snyder <wsnyder@wsnyder.org>
Date:   Fri Oct 23 18:23:02 2020 -0400

    Fix multiplication in multidimensional AUTOINST output (#1698).
    
    * verilog-mode.el (verilog-read-sub-decls-expr):
    Fix multiplication in multidimensional AUTOINST output (#1698). Reported by alanamckee.

```

iperryman-xeda, 2020-11-04:
Actually it works fine in emacs 26.1, it is only the XEmacs that it fails to compile.

I was able to run make test_emacs successfully:

Tests Passed on GNU Emacs 26.1 (build 1, x86_64-redhat-linux-gnu, GTK+ Version 3.22.30)
 of 2019-05-11
9.28user 1.83system 0:15.00elapsed 74%CPU (0avgtext+0avgdata 67276maxresident)k
1008inputs+664outputs (0major+193989minor)pagefaults 0swaps

wsnyder, 2020-11-05:
I presume you're using GNU Emacs so are happy as-is.

xemacs 21.5 is still in beta and doesn't seem to even build on Ubuntu 20.04.  If you or someone would like to debug this, we'd be happy to take the fix.

iperryman-xeda, 2020-11-05:
Correct. I am only using GNU Emacs.  We have 1 user who prefers XEMacs.
I'll look into it. I presume CentOS is not in your regressions.

Ian

wsnyder, 2020-11-07:
Thanks, based on the pull I made a different fix so 21.4 would still work.

closed 2020-11-07 by wsnyder

monnier, 2021-04-12:
The fix looks odd since `(eval "blabla")` will just return `"blabla"`.  AFAICT the error is simply due to passing the arguments to `member` in the wrong order:

```
@@ -2077,8 +2081,7 @@ find the errors."
       (if (boundp 'compilation-error-regexp-systems-alist)
 	  (if (and
 	       (not (equal compilation-error-regexp-systems-list 'all))
-               ;; eval required due to bug1700, XEmacs otherwise errors on compile
-	       (not (eval "(member compilation-error-regexp-systems-list 'verilog)")))
+	       (not (member 'verilog compilation-error-regexp-systems-list))
 	      (push 'verilog compilation-error-regexp-systems-list)))
       (if (boundp 'compilation-error-regexp-alist-alist)
 	  (if (not (assoc 'verilog compilation-error-regexp-alist-alist))
```