veripool/verilog-mode #1703: Advanced Indentation

NickSica, 2020-12-07:
I don't know if there is a correct place to ask this but is there a way to set more fine grained indentation behavior?
For instance I want the following:
`module adder(
  input logic a,
  input logic b,
  output logic y,
);`
with the ports indented once and the closing parentheses on the same line if I were to put the first input on the previous line I would want it like the following:
`module adder(input logic a,
                         input logic b,
                         output logic y,
);`

I realized that there is an indent lists variable but disabling that causes them to not line up like the second example. If that functionality doesn't exist could you point me at where to look to implement said functionality? Thank you!

wsnyder, 2020-12-07:
I can see how that's nicer, I don't believe there is a setting for this, but improvements would be appreciated.  Perhaps  @kaushalmodi or @acr4 can comment.

acr4, 2020-12-07:
I have two thoughts here...

1) Enabling user-customized / fine-grain indentation would be pretty
slick.  There are definitely times when I dislike the default format of
various utilities.  Maybe that could be done by looking for a user-defined
call-back at the entry of each formatting routine?  But, a lot of the
verilog-mode code is, well, ugly.  Sorry, it just is.  I certainly am not
blameless here.  Most of this code was written by hardware engineers, and
we're a particularly sloppy bunch when it comes to writing software :).
There's often not a very great delineation between parsing (what language
construct am I in?) and modifying (indenting said construct).

2) Adding additional complexity is going to further slow down some areas
that can be pretty slow already.  Adding additional features for
user-specific customization is likely going to expose all sorts of ugly
corner cases.  It will be a support nightmare (even if it's just
determining _who_ is at fault -- verilog-emacs or user-code).  It's a
really slippery slope too.  Who has time to implement said features?  Who
has time to support it?

Saying "no" to every feature request just means the code atrophies.  Fewer
and fewer people understand it.  It dies a slow death.  Saying "yes" to
every feature request results in unmanageable code with too many
corner-cases, too many places for bugs to hide, hurts performance, etc.
Finding the balance between a technical asset and a technical debt is
challenging.

As neat as this sounds, I don't think the language parsing and indention
code is currently in the right place to support this sort of enhancement.
I unfortunately do not have the time to develop/test/support such a change,
nor do I have the bandwidth to begin refactoring or rearchitecting the
existing code.  It has, through necessity, grown rather organically from
Verilog-95 through to SV-2012 syntax.  Along with that comes the lack of
agility in many areas of the requisite code base.

NickSica, 2020-12-13:
If you could point me in the direction of where to look for this sort of thing, I could probably get something working if you are leaning towards allowing something like this. Otherwise maybe there is an easier way around this problem?

bluewww, 2020-12-17:
I personally have some hacks in my `verilog-mode.el` to allow for better matching with lowRISC's style guide. I'd also really like better indendation support in mainline verilog-mode. I can see if something from my changes can be re-used.

ruelparent, 2020-12-17:
The signals are aligned with the instance opening '('. so if you put it on the next line it will line up like your first example:

module adder
  (input logic a,
   input logic b,
   output logic y,
);

it may look strange but it works with current indenting

closed 2022-05-12 by wsnyder