veripool/verilog-mode #1710: emacs/verilog-mode vs LSP?

unfrostedpoptart, 2021-01-16:
Hi.

I've read some stuff about LSP and see there's some Verilog support but I don't really understand what exactly LSP would do vs using verilog-mode.  Can you please give an explanation of the similarities and differences and if there's any advantage - and plan - to have verilog-mode use LSP?

Thanks!

 David

wsnyder, 2021-01-17:
What's LSP?  That probably tells you much of what you were asking ;)

unfrostedpoptart, 2021-01-17:
https://github.com/emacs-lsp/lsp-mode

https://en.wikipedia.org/wiki/Language_Server_Protocol?wprov=sfti1

wsnyder, 2021-01-17:
There's no current plans to use LSP or make major overhauls of verilog-mode, though welcome others to consider improvements/contributions.

closed 2021-01-17 by wsnyder