veripool/verilog-mode #1711: intendiation wrong if within and ifdef else

joachimlechner, 2021-01-19:
Hello,

in the following code the intendiation is wrong - see ISSUE comment.
Please also find attached a minimum example.

BR, Joachim

```
`ifdef TEST
   foo1 bar
`else
     foo2 bar // ISSUE: indentiation wrong, should be like for foo1 bar !
`endif
       (/*AUTOINST*/);
```

[ifdef_instance.tar.gz](https://github.com/veripool/verilog-mode/files/5836824/ifdef_instance.tar.gz)

wsnyder, 2021-01-23:
Verilog-mode can't really handle all the possible ifdef combinations so this is unlikely to get fixed; furthermore it will confuse the AUTOs.  I suggest instead this syntax:

```
`ifdef TEST
 `define BAR_INST foo1
`else
 `define BAR_INST foo2
`endif
   `BAR_INST bar
     (/*AUTOINST*/
      // Inputs
      .signal                         (signal));

endmodule

// Local Variables:
// eval:(verilog-read-defines)
// End:
```

closed 2021-01-23 by wsnyder

acr4, 2021-01-23:
I agree with Wilson.  Stylistically this s is also a lot easier to read. It sets up the intent (pick a different module type based on configuration) and separates it from the boring language mechanics.

joachimlechner, 2021-01-25:
Thanks for the suggested syntax that works also for me.