veripool/verilog-mode #1712: autowire creates wrong width definition

joachimlechner, 2021-01-19:
Hello,

in the following code the automatic wire contains a wrong width definition. See ISSUE.
(fyi. This worked in a previous version from ~1-2years, found after updating to the latest code here).
Please also find attached a minimum example.

BR, Joachim

[parameter_auto_wire.tar.gz](https://github.com/veripool/verilog-mode/files/5836840/parameter_auto_wire.tar.gz)

```
   parameter WIDTH = 16;

   // ISSUE: in autowire below [WIDTH*0-1:0] should have been [WIDTH*2/8-1:0]

   /*AUTOWIRE*/
   // Beginning of automatic wires (for undeclared instantiated-module outputs)
   wire [WIDTH*0-1:0]   signal;                 // From subtestout of subtestout.v
   // End of automatics

   subtestin #(.WIDTH(WIDTH)) subtestin
     (/*AUTOINST*/
      // Inputs
      .signal                           (signal[WIDTH*2/8-1:0]));
```

wsnyder, 2021-01-23:
Thanks for the good report. As you probably guessed it was assuming X/Y had no remainder, so 2/8=0.  Fixed.

closed 2021-01-23 by wsnyder

joachimlechner, 2021-01-25:
You are welcome. Thanks for fixing!