veripool/verilog-mode #1713: Previously Working AUTOOUTPUT, AUTOINPUT does not expand anymore with no errors - How to debug

engrvns, 2021-01-27:
Hi verilog-mode team

This is a weird issue, 
A previously working RTL file with AUTOOUTPUT, AUTOINPUT does not expand the inputs and outputs.
All other AUTOs are working fine.
Killing the AUTOs or using AUTOARGs instead non of them create the ports.

NO errors in the message buffer.
I am on the latest verilog-mode.

Any clues on how to debug this?

Thanks

engrvns, 2021-01-27:
Basically copied the file into a new file and killed the autos and expanded it. Worked
No other difference...
Crude but works.
 It would interesting to know how to debug why the expansion did not work in the first file.

wsnyder, 2021-01-27:
Strange, sorry I don't have any ideas.

All I could think of is some strange syntax or control character in the file, which when you copied didn't come along.

closed 2021-01-27 by wsnyder