veripool/verilog-mode #1717: indent issue after double comment or comment in input signal line ?

joachimlechner, 2021-02-24:
Hello,

In the following code test1 is indented wrong.
It seems **wire** is aligned to **test3** from before.
I am using the latest verilog-mode version.

```
module test
  (input wire test5 // bla
   /*AUTOARG*/);

`include "bla.vh"

   reg        test4;

   wire       test3;

   //  // FIXXME
              wire       test1 = test2; // wrong indent
   wire                  test = test;

endmodule
```

Sorry I could not break it down further.

BR, Joachim

wsnyder, 2021-02-26:
Looks like verilog-continued-line-1 is not handling this.  Needs more debug if you or someone gets a chance.

gmlarumbe, 2022-07-02:
Hi,

Commit 4730c15 fixes the issue with GNU Emacs 28.1. 

Can you test if everything still works fine with this patch before merging into master?

Thanks!

closed 2022-07-12 by gmlarumbe