veripool/verilog-mode #1719: Incorrect indentation when using nested curly brackets

leventyakay, 2021-03-25:
When double curly brackets are used to do bit replication and/or concatenation, indentation breaks starting with next line.
Example:

always_ff @(posedge clk or negedge rst_n)
  if (~rst_n)
    begin
    a <= {(5){1'b0}};
  // auto-indentation fails
  a <= 1;
  end

always_ff @(posedge clk or negedge rst_n)
  if (~rst_n)
    begin
    a <= {5{1'b0}};
    // auto-indentation works
    a <= 1;
    end

always_ff @(posedge clk or negedge rst_n)
  if (~rst_n)
    begin
    a <= {{1'b0,1'b0}};
  // auto-indentation fails
  a <= 1;
  end

wsnyder, 2021-03-26:
Somewhat surprising, but confirmed.  Note indentation bugs are in mostly in only-fixed-by-contribution mode, so if you could look into a pull request to fix this, it would be appreciated.

acr4, 2021-03-26:
I can also confirm this issue still exists.  I tried to clean up some of
the {{ and }} quirks years ago, and never managed to quite beat it into
submission.  That code is ... squirrely.

leventyakay, 2021-03-26:
Unfortunately I am no expert in lisp. When I have some spare time, I'll see if I can do a fix. I didn't use github before though.

wsnyder, 2021-09-17:
@punzik, appreciate your fix for this!

closed 2021-09-17 by wsnyder