veripool/verilog-mode #1720: typedef signal name recognition problem

AlphaLyrae0, 2021-04-07:
It turned out when signal names have the same suffix as type name of  typedef which is specified as local variable, then, those signals are not recognized in AUTOINST.
I'd appreciated it if this problem could be resolved in the future release.

```
module example_sub(
        input my_t           aaa_t, // NG
        input my_pkg::my_t   bbb_t, // NG
        input my_t           aaa  , // OK
        input my_pkg::my_t   bbb  , // OK
       	input            a ,
       	output x
);
endmodule
// Local Variables:
// verilog-typedef-regexp:"_t$"
// End:
```
```
module example_top 
(
	/*AUTOINPUT*/
	// Beginning of automatic inputs (from unused autoinst inputs)
	input		a,			// To i_sub of example_sub.v
	input my_t	aaa,			// To i_sub of example_sub.v
	input my_pkg::my_t bbb,			// To i_sub of example_sub.v
	// End of automatics
	/*AUTOOUTPUT*/
	// Beginning of automatic outputs (from unused autoinst outputs)
	output		x,			// From i_sub of example_sub.v
	// End of automatics
);
   example_sub  i_sub(
		      /*AUTOINST*/
		      // Outputs
		      .x		(x),
		      // Inputs
		      .aaa		(aaa),
		      .bbb		(bbb),
		      .a		(a));
endmodule
// Local Variables:
// verilog-typedef-regexp:"_t$"
// End:
```

Thansk.

wsnyder, 2021-04-07:
Unfortunately this is fundamental to how Verilog-Mode does its parsing, so isn't something that will be fixed.  You'll need to come up with a verilog-typedef-regexp that does not match the signal names, or better adopt a methodology which does not allow signals to have names that match the regexp (e.g. end in _t), or manually declare the signals.

closed 2021-04-07 by wsnyder

AlphaLyrae0, 2021-04-07:
And also,