veripool/verilog-mode #1721: Spelling mistake in verilog-mode flag causes others to be ignored

udif, 2021-05-04:
Just found this one by pure chance:
One of my files had:
```
// Local Variables:
// verilog-library-directories:("." ... )
// verilog-library-extensions:(".v")
// verilog-auto-ignire-concat:t
// End:
```
(Notice the spelling mistake 'ignire')
This prevented verilog-mode from finding a module instance reference, as if the `verilog-library-directories` variable did not exist.

Fixing the spelling mistake above to:
```
// verilog-auto-ignore-concat:t
```
caused the `verilog-library-directories` variable above to be active again.

wstucker, 2021-05-04:
@udif By default, I would have expected this to be caught by the unsafe local variables check:

```
The local variables list in example.v
contains values that may not be safe (*).

Do you want to apply it?  You can type
y  -- to apply the local variables list.
n  -- to ignore the local variables list.
!  -- to apply the local variables list, and permanently mark these
      values (*) as safe (in the future, they will be set automatically.)

    verilog-library-directories : ("." \.\.\.)
    verilog-library-extensions : (".v")
  * verilog-auto-ignire-concat : t
```

Can you check the value of ` enable-local-variables`?

udif, 2021-05-04:
Where was that supposed to be defined?
I'm running this exclusively outside of emacs in batch mode, and I'm as far as possible from being an emacs expert (verilog-mode in batch-mode is the only reason emacs is installed on my machine).
```
emacs --batch file.v -l $HOME/git/verilog-mode/e/verilog-mode.el -f verilog-batch-auto
```

You can find the following (line 5598) inside verilog-mode.el itself:
```
 5586 (defun verilog-batch-execute-func (funref &optional no-save)
 5587   "Internal processing of a batch command.
 5588 Runs FUNREF on all command arguments.
 5589 Save the result unless optional NO-SAVE is t."
 5590   (verilog-batch-error-wrapper
 5591    ;; Setting global variables like that is *VERY NASTY* !!!  --Stef
 5592    ;; However, this function is called only when Emacs is being used as
 5593    ;; a standalone language instead of as an editor, so we'll live.
 5594    ;;
 5595    ;; General globals needed
 5596    (setq make-backup-files nil)
 5597    (setq-default make-backup-files nil)
 5598    (setq enable-local-variables t)
```

The name of this function, `verilog-batch-execute-func` suggests to me it is used in batch-mode.

wsnyder, 2021-05-04:
Yes, that is used in batch mode.  The common use of lisp in these end functions requires that.  Emacs itself does these local variable blocks and doesn't have any way I know of to deal with the misspellings.  So unfortunately I don't think there's anything verilog-mode can do to make this better.

I'm going to close, but if better ideas come up feel free to follow up.

closed 2021-05-04 by wsnyder