veripool/verilog-mode #1722: AUTO_TEMPLATE - regex for adding instance number times 8 to port number

longhorngeek, 2021-05-10:
Hello Wilson,

I can't figure out the regex in AUTO_TEMPLATE to get the following output for 2 consecutive instances that each have 8 ports. I can make it work in bus form, but I need the signals to remain separate - not in a bus.

This puts it into a bus
/*
   testblock AUTO_TEMPLATE
     (
      .out_@         (this_is_out_[(@*8)+\1]),
     );
*/

   testblock testblock_0
     (/*AUTOINST*/
      // Outputs
      .out_0          (this_is_out_0),
      .out_1           (this_is_out_1),
      .out_2          (this_is_out_2),
      .out_3          (this_is_out_3),
      .out_4          (this_is_out_4),
      .out_5          (this_is_out_5),
      .out_6          (this_is_out_6),
      .out_7          (this_is_out_7),
   /*

   testblock testblock_1
     (/*AUTOINST*/
      // Outputs
      .out_0          (this_is_out_8),
      .out_1           (this_is_out_9),
      .out_2          (this_is_out_10),
      .out_3          (this_is_out_11),
      .out_4          (this_is_out_12),
      .out_5          (this_is_out_13),
      .out_6          (this_is_out_14),
      .out_7          (this_is_out_15),
   /*

Thanks and regards,

Jim

wsnyder, 2021-05-10:
The text in the template is what will come out (after substitution)

         .out_@         (this_is_out_[(@*8)+\1]),

Will thus make out_[...], so the simple answer is don't put the brackets into the template.  But I think your larger question is probably how to get the @ to become a string.  To do this you need to use lisp to calculate and convert it to a string.

      /* a AUTO_TEMPLATE(
        .out_@         (this_is_out_@"(number-to-string (+ (* @ 8) \1))"),
       ) */

closed 2021-05-10 by wsnyder

longhorngeek, 2021-05-14:
Thank you! It works exactly as intended.