veripool/verilog-mode #1725: Verilog mode treat typedef as output port

futurehome, 2021-06-17:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (You may want to attach output from 'M-x verilog-submit-bug-report')
The Emacs version is 24.3.1, and the example file is test.txt, and the output is in the pic below. The tool just treats OUT_t as an output port, not a type. Any missing settings? Thanks, Alex
[test.txt](https://github.com/veripool/verilog-mode/files/6667000/test.txt)
[image]

Can you please check your github name is set to your real name (click on your avatar icon in upper right, then "settings" then "Name")?

wsnyder, 2021-06-17:
I suspect it's because you shouldn't backslash the quotes in a variable list.

           // verilog-typedef-regexp: "_t$"

closed 2021-06-17 by wsnyder

futurehome, 2021-06-17:
It works, thanks.

Another question, I want to keep the pin list in declaration order, and I add the 2 lines below in each of my files, but there is no effect, anything missing in my setting? 
       // verilog-auto-arg-sort: nil
       // verilog-auto-inst-sort: nil
[test.txt](https://github.com/veripool/verilog-mode/files/6667233/test.txt)
[InstModule.txt](https://github.com/veripool/verilog-mode/files/6667234/InstModule.txt)

wsnyder, 2021-06-17:
This works for me, note it only sorts within a category, inputs and outputs will always be separated if that's what you think is wrong.

futurehome, 2021-06-18:
I see. 
Instead of putting Local Variable in every file, is there any way to set globally?

Thanks,
Alex

wsnyder, 2021-06-18:
You can put it in a .emacs or site-wide file.  See e.g. https://www.math.utah.edu/docs/info/emacs-lisp-intro_17.html

futurehome, 2021-06-21:
I added the line below in .emacs, and removed Local Variables in all my files, but seems the OUT_t still be treated as an output port, not a type.

(setq verilog-typedef-regexp "_t$")

wsnyder, 2021-06-21:
Maybe

(setq-default verilog-typedef-regexp "_t$")

futurehome, 2021-06-21:
I know the root cause, because I run emacs in batch mode, it will skip loading the init file. 
Thanks for the help.