veripool/verilog-mode #1726: Match only alphabet in cell name

imgod2u, 2021-06-21:
I'm trying the following template:

    .\(.*\) (@"(verilog-string-replace-matches \\"^\\\\([a-z]*\\\\)_\\\\w*$\\" \\"\\\\1\\" nil nil (downcase vl-cell-name))"_\1),

I want to prepend the first alphabetic sequence ("pdd" in "pdd_inst") to each signal name.

However, the string [a-z] doesn't seem to match and my signal always ends up:
    <cell-name>_sig

Any help with the syntax here would be greatly appreciated. I've tried a few variants:
    .\(.*\) (@"(verilog-string-replace-matches \\"^\\\\(\\\\[a-z\\\\]*\\\\)_\\\\w*$\\" \\"\\\\1\\" nil nil (downcase vl-cell-name))"_\1),
    .\(.*\) (@"(verilog-string-replace-matches \\"^\\\\(\\\\[a-z\\\\]\\\\+\\\\)_\\\\w*$\\" \\"\\\\1\\" nil nil (downcase vl-cell-name))"_\1),

With no success.

wsnyder, 2021-06-21:
Don't escape the brackets, only parenthesis.  Also I would avoid using the \w escape, use e.g. [a-zA-Z0-9_]

imgod2u, 2021-06-21:
Thanks, the following:

    .\(.*\) (@"(verilog-string-replace-matches \\"^\\\\([a-z]\\\\)_[a-zA-Z0-9_]$\\" \\"\\\\1\\" nil nil (downcase vl-cell-name))"_@"(downcase vl-name)[]"),

Produces:

    wholeCellName_sigName

So it seems [a-z] isn't working....

imgod2u, 2021-06-21:
Thanks a lot! That worked.

closed 2021-06-21 by imgod2u