veripool/verilog-mode #1727: Have AUTOOUTPUT ignore signals that have been declared

imgod2u, 2021-06-23:
I have signals declared in my module like such:

    module top_module(
      /*AUTOINPUT*/
      /*AUTOOUTPUT*/
    )

    logic sig1;
    logic [2:0] sig2;

    my_module dut (
      /*AUTOINST*/
      sig1(sig1);
      sig2(sig2);
    );
    endmodule

sig1 and sig2 are signals I'd like to leave dangling (and the testbench will drive it through hierarchical references).

Problem is, /*AUTOOUTPUT*/ will see that it isn't the input of another module and declare the output automatically.

Is there a way to have AUTOOUTPUT ignore signals that are declared?

wsnyder, 2021-06-23:
See here please https://github.com/veripool/verilog-mode/blob/master/FAQ.rst#how-do-i-remove-outputs-from-autooutput

closed 2021-06-23 by wsnyder

imgod2u, 2021-06-23:
Hmm, that isn't quite what I'm looking for though. I'd have to list the signals I want ignored twice (once for the dangling wire declaration and once in the ignore list or fake output declaration). I take it there's no way to get it to see dangling wire declarations as a sink?

wsnyder, 2021-06-23:
Perhaps this

```
`ifdef NEVER   // For autos as driven by testbench
    assign out = FROM_TB;
`endif
```

wsnyder, 2021-06-23:
Or I think you want the other direction:
`ifdef NEVER   // For autos as driven by testbench
    tbdriven  tbdriven (// Inputs
                                   .out);
`endif