veripool/verilog-mode #1731: /*AUTOWIRE*/ is ignoring existing declaring of typedef objectss

afuller525, 2021-07-10:
I am attempting to use user-defined struct objects on the port list of a submodule. I discovered that /*AUTOWIRE*/ is failing to notice if an intermediate bus is already defined if it is of a user-defined type. This leads to the bus being added to the automatic wire list unnecessarily, and then the compiler gives a redeclaration error.

Below is a very contrived and simplified example. In the top module, the object "b" gets added to the automatic wires list even though it is already declared in this module scope.

<<<file: top.v>>>

module top (c, a);
   
    typedef struct {
	logic	x;
	logic	y;
    } mystruct_t;

    /*AUTOOUTPUT*/
    // Beginning of automatic outputs (from unused autoinst outputs)
    output mystruct_t	c;			// From inst1 of submod.v
    // End of automatics

    /*AUTOINPUT*/
    // Beginning of automatic inputs (from unused autoinst inputs)
    input mystruct_t	a;			// To inst0 of submod.v
    // End of automatics

    // Wire declarations
    mystruct_t b;
    
    /*AUTOWIRE*/
    // Beginning of automatic wires (for undeclared instantiated-module outputs)
    mystruct_t		b;			// From inst0 of submod.v
    // End of automatics

    /* submod AUTO_TEMPLATE (
     .out	(b[]),
     .in	(a[]),
     ); */    
    submod	inst0
	(// Outputs
	 /*AUTOINST*/
	 // Outputs
	 .out				(b),			 // Templated
	 // Inputs
	 .in				(a));			 // Templated

    /* submod AUTO_TEMPLATE (
     .out	(c[]),
     .in	(b[]),
     ); */    
    submod	inst1
	(/*AUTOINST*/
	 // Outputs
	 .out				(c),			 // Templated
	 // Inputs
	 .in				(b));			 // Templated
    
endmodule
// Local Variables:
// verilog-typedef-regexp: "_t$"
// End:

<<<file: submod.v>>>

module submod (out, in);
    typedef struct {
	logic	x;
	logic	y;
    } mystruct_t;

    output mystruct_t	out;
    input mystruct_t	in;

    assign		out = ~in;

endmodule
// Local Variables:
// verilog-typedef-regexp: "_t$"
// End:

Thanks for your help.

wsnyder, 2021-07-11:
I don't see the AUTOWIRE declaring mystruct_t.  Are you sure you are using the most recent version of verilog-mode?

afuller525, 2021-07-11:
I apologize, it looks like my company is using a very old version (840). I tested a local override with the newest version and the problem is resolved.

Thanks.

closed 2021-07-11 by afuller525