veripool/verilog-mode #1734: Can a lisp expression in an AUTO block use a parameter from a package?

sjalloq, 2021-07-27:
I may well be trying to be too clever here but I wanted to perform a lisp expansion in a port connection along the lines of:

```verilog
/* module AUTO_TEMPLATE (
 .signal_@ (some_bus[@"(+ \1 OffsetParameterInPackageFile)"]),
);
*/
```

I've got `verilog-library-directories` set and I tried `verilog-auto-read-includes:t` but I get an error that the Symbol's value is void.  Does this only work with `include files and defines?

wsnyder, 2021-07-27:
You need to prepend "vh-" to get parameter values.  See e.g. https://github.com/veripool/verilog-mode/blob/master/tests_ok/autoinst_param_value.v

wsnyder, 2021-07-27:
P.S. Feel free to reopen or post followups if I misunderstood your issue!

closed 2021-07-27 by wsnyder

sjalloq, 2021-07-27:
Still doesn't work.  Here's an example.  I've tried to make it easier by having the parameter in the same module for now but I get a different error.

1.  If I define the parameter in module test below, I get the error, "Wrong type argument: number-or-marker-p, "9"".
2. If I import the package only, I get, "Symbolâ€™s value as variable is void: vh-MyParam".

```verilog
module sub(/*AUTOARG*/);
  output logic signal_1;
endmodule

module test
  import test_pkg::*;
  (/*AUTOARG*/);

  parameter MyParam = 7;

  /*AUTOOUTPUT*/

  /* sub AUTO_TEMPLATE (
   .signal_@ (signal[@"(+ 1 vh-MyParam)"]),
   );
   */
  sub u_sub (/*AUTOINST*/);

endmodule

// Local Variables:
// verilog-library-extensions:(".v" ".sv")
// verilog-auto-read-includes:t
// verilog-library-directories:(".")
// End:
```

sjalloq, 2021-07-27:
Ah, issue 1 above is solved by using `@"(+ 1 (string-to-number vh-MyParam))"` but I haven't seen any examples using string to number conversion...

wsnyder, 2021-07-27:
Correct.   If you were writing lisp you'd then after adding need a number-to-string, but verilog-mode templates do that for you when it sees a number.

The vh-* values are file local variables in the appropriate file.  So if you need the value from "foo.vh" you need to do (untested):

      (save-excursion (set-buffer "foo.vh") vh-MyParam))

Said another way, vh- doesn't do any imports nor understand what is really a package.

sjalloq, 2021-07-27:
On the other issue, I just found that I can do `// eval:(verilog-read-defines "test_pkg.sv")` but that only works if my parameter is not declared as an int.

So `parameter int MyParam = 10` doesn't work but `parameter MyParam = 10` does.

wsnyder, 2021-07-28:
Looks like read-defines was never improved to handle SystemVerilog parameters with types.  I pushed a fix for this to the verilog-mode here, please give it a try, thanks.