veripool/verilog-mode #1736: `end // UNMATCHED !!` after `analog begin` in Verilog-A

danmcmahill, 2021-08-19:
I often times use verilog-mode for editing Verilog-A code due to syntax highlighting and indentation.  One minor glitch I hit on every file is the auto generated comment after the closing `end` for an `analog begin` block fails.

Here is an example.  As a note, every Verilog-A module will contain the `analog` block.  Verilog-AMS users will encounter this as well.

```
module mymodule();
   analog begin
      // contents of the module goes here
   end // UNMATCHED !!
endmodule // mymodule
```
To reproduce, just delete the ` // UNMATCHED !!` and press enter and see it come back.  What would be preferred is ` // analog` instead of `// UNMATCHED !!` so I typically just manually make that change now in each Verilog-A file.

With a push in the right direction in the verilog-mode.el code I may be able to create a patch but I'm a little lost as to where to start.

wsnyder, 2021-08-19:
I speculate you can treat it like e.g. always_comb, that is add "analog" to verilog-no-indent-begin-re and perhaps other spots.  If you could please make a pull request with that change, a test and test_ok file and give it a try, it would be appreciated.

danmcmahill, 2021-09-01:
Thanks!  Submitting PR now.

gmlarumbe, 2022-07-03:
PR: #1738 

Thanks!

closed 2022-07-03 by gmlarumbe