veripool/verilog-mode #1739: AUTO_TEMPLATE that applies only to a specific instance

imgod2u, 2021-09-01:
I have multiple instances of the same module, for example:
```
/* my_module AUTO_TEMPLATE "^\(.*\)" (
.\(.*\) (@_\1),
);
*/
my_module u0(
  /*AUTOINST*/
);
my_module u1(
  /*AUTOINST*/
);
my_module u2(
  /*AUTOINST*/
);
```

However, instance u2 needs more specific connections that's different than the others. For example, I want port2 of u2 tied to 1'b0 but u0 and u1 to tie it to u0_port2 and u1_port2 respectively.

Is there a way to make specific template rules based on instance name?

wsnyder, 2021-09-01:
For your case it sounds like you can just list the exceptions.

```
/* my_module AUTO_TEMPLATE "^\(.*\)" (
.\(.*\) (@_\1),
);
*/
my_module u0(
  /*AUTOINST*/
);
my_module u1(
  // Outputs
  .port2('0),
  /*AUTOINST*/
);
```

Another alternative is to put (and repeat the identical rules)

```
/* my_module AUTO_TEMPLATE "^\(.*\)" (
.\(.*\) (@_\1),
);
*/
my_module u0(
  /*AUTOINST*/
);
/* my_module AUTO_TEMPLATE "^\(.*\)" (
.\(.*\) (@_\1),
);
*/
my_module u1(
  /*AUTOINST*/
);
```

wsnyder, 2021-09-01:
For your case it sounds like you can just list the exceptions.

```
/* my_module AUTO_TEMPLATE "^\(.*\)" (
.\(.*\) (@_\1),
);
*/
my_module u0(
  /*AUTOINST*/
);
my_module u1(
  // Outputs
  .port2('0),
  /*AUTOINST*/
);
```

Another alternative is to put a template for each (and repeat the identical rules)

```
/* my_module AUTO_TEMPLATE "^\(.*\)" (
.\(.*\) (@_\1),
);
*/
my_module u0(
  /*AUTOINST*/
);
/* my_module AUTO_TEMPLATE "^\(.*\)" (
.port2('0),
.\(.*\) (@_\1),
);
*/
my_module u1(
  /*AUTOINST*/
);
```

closed 2021-09-01 by wsnyder