veripool/verilog-mode #1740: "System verilog attribute on pin" autoinsted by  AUTO_TEMPLATE

13771915047, 2021-09-07:
Hi, 
SystemVerilog attribute on a pin within a cell instantiation in the RTL sets that attribute
on the corresponding instance pin in the HDL Compiler database.
RTL:
PLL UPLL1 (.REFCLK(CLK1), .FDBCK(CLK1_FDBCK),
(* my_pll_mult = 2 *) .CLKOUT2(PLLCLK1_X2),
(* my_pll_mult = 4 *) .CLKOUT4(PLLCLK1_X4));
PLL UPLL2 (.REFCLK(CLK2), .FDBCK(CLK2_FDBCK),
(* my_pll_mult = 2 *) .CLKOUT2(PLLCLK2_X2),
(* my_pll_mult = 4 *) .CLKOUT4(PLLCLK2_X4));

And I would like to request could these attributes be autoinsted by AUTO_TEMPLATE?

for example: 
set AUTO_TEMPLATE like this:
/*  module_test  AUTO_TEMPLATE (
  (* clk_freq = 125 ,pin_type="clk"*)  .clk  (clk_sys),
  (* pin_type = "rst" *)  .rstb  (rstb_mac),
);*/

expected autoinst result:
     module_test test_inst(/*autoinst*/
  (* clk_freq = 125 ,pin_type="clk"*)  .clk  (clk_sys),
  (* pin_type = "rst" *)  .rstb  (rstb_mac));

Thanks for reading!

wsnyder, 2021-09-07:
Sorry, there's no plans to support this, as it's very uncommonly used, and painful to implement.  I'd suggest not using autos for this example.

closed 2021-09-07 by wsnyder

13771915047, 2021-09-08:
Hi, get it, thank you all the same!