veripool/verilog-mode #1741: verilog-auto-inst-sort does not work

zhanjf, 2021-09-14:
hi, I don't want to reorder the port list and try to use the verilog-auto-inst-sort.
module foo(/*AUTOARG*/);

input clk;
output b;
input rstn;
output a;

endmodule // top_md

module top_md(/*AUTOARG*/);
input           din1;
input [1:0]     din2;

output          dout1;
output [1:0]    dout2;

/*AUTOREG*/
/*AUTOWIRE*/

wire            din_a;
wire            din_b;

always@(/*AUTOSENSE*/)
    dout1 = din1 | din2[1];

assign din_a = din2[0];
assign din_b = din2[1];

assign dout2[1:0] = dout_a[1:0];

/*
foo AUTO_TEMPLATE(
);
 */

foo u_foo(/*AUTOINST*/);

endmodule // top_md

//Local Variables:
//verilog-auto-inst-sort:nil
//verilog-auto-arg-sort:nil
//verilog-auto-inst-param-value:t
//indent-tabs-mode:nil
//End:

and the result is 

module foo(/*AUTOARG*/
   // Outputs
   b, a,
   // Inputs
   clk, rstn
   );

input clk;
output b;
input rstn;
output a;

endmodule // top_md

module top_md(/*AUTOARG*/
   // Outputs
   dout1, dout2,
   // Inputs
   din1, din2
   );
input           din1;
input [1:0]     din2;

output          dout1;
output [1:0]    dout2;

/*AUTOREG*/
// Beginning of automatic regs (for this module's undeclared outputs)
reg                     dout1;
// End of automatics
/*AUTOWIRE*/
// Beginning of automatic wires (for undeclared instantiated-module outputs)
wire                    a;                      // From u_foo of foo.v
wire                    b;                      // From u_foo of foo.v
// End of automatics

wire            din_a;
wire            din_b;

always@(/*AUTOSENSE*/din1 or din2)
    dout1 = din1 | din2[1];

assign din_a = din2[0];
assign din_b = din2[1];

assign dout2[1:0] = dout_a[1:0];

/*
foo AUTO_TEMPLATE(
);
 */

foo u_foo(/*AUTOINST*/
          // Outputs
          .b                            (b),
          .a                            (a),
          // Inputs
          .clk                          (clk),
          .rstn                         (rstn));

endmodule // top_md

//Local Variables:
//verilog-auto-inst-sort:nil
//verilog-auto-arg-sort:nil
//verilog-auto-inst-param-value:t
//indent-tabs-mode:nil
//End:

it seems  verilog-auto-inst-sort does not work?

wsnyder, 2021-09-14:
Changing to verilog-auto-inst-sort:t does change the order:

```
           // Outputs
-          .b                            (b),
           .a                            (a),
+          .b                            (b),
           // Inputs
```

Maybe you're using a older version or something?

zhanjf, 2021-09-15:
I just  git the file and make it, I think it's new one, right?  Wsnyder Could you please try my example with //verilog-auto-inst-sort:nil?

wsnyder, 2021-09-15:
nil doesn't do the sort, the differences between the two (nil vs t) are shown above.

zhanjf, 2021-09-15:
So, what's wrong with my verilog-mode? I updated the verilog-mode.el. Can i  add the verilog-mode version information in the verilog file?

wsnyder, 2021-09-15:
See if "M-x load-file <path_to_verilog_mode.el>" works.  If so perhaps you aren't putting the file in a place your Emacs load-file path can find it.  If it doesn't work, I don't know what's wrong ;)

zhanjf, 2021-09-15:
Thanks for you help. I checked the version of emacs, it's really the old one . But very strange it cannot work even if I updated it. 

You are using verilog-mode 2021-09-01-b6d4104-vpo-GNU

For new releases, see https://www.veripool.org/verilog-mode

For frequently asked questions, see https://www.veripool.org/verilog-mode-faq.html

To submit a bug, use M-x verilog-submit-bug-report

zhanjf, 2021-09-15:
Hi, wsnyder, what's your emacs version? thanks!

wsnyder, 2021-09-16:
Does "M-x describe-variable verilog-auto-inst-sort" show the right value?

Are you sure it's not working correctly - you get results different than I posted?  (Versus what you personally think "sorted" might mean ;)

closed 2021-09-17 by wsnyder