veripool/verilog-mode #1751: AUTOINST extra space for "// Templated" on last port of instance

fnJeff, 2021-11-11:
This has been a minor annoyance of mine since I first started using the verilog mode 20+ years ago.  I actually tried to fix it once myself.  I thought I found the related code, but with zero LISP knowledge, I was never able to modify the code in a way that fixed it no matter how much I experimented.  This might be a super easy fix for someone more familiar with LISP.

Below you can see the extra space before the last "// Templated" comment.  I am fairly certain this is due to the extra ")" that shows up only on that last line.  It isn't accounted for in the calculation of the number of spaces to insert to align the comments.

If anyone can fix this... my OCD would greatly appreciate it.  :)

```
module debug_auto_inst
   (input  logic [3:0] a,
    input  logic [3:0] b,
    output logic [4:0] c
    );
   assign c = a + b;
endmodule

module top;
   logic signed [3:0] a_int;
   logic signed [3:0] b_int;

   /*AUTOLOGIC*/
   // Beginning of automatic wires (for undeclared instantiated-module outputs)
   logic [4:0]          c_int;                  // From debug_auto_inst of debug_auto_inst.v
   // End of automatics

   /*debug_auto_inst AUTO_TEMPLATE (
       .\([abc]\)                      (\1_int[]),
    );*/
   debug_auto_inst debug_auto_inst
      (/*AUTOINST*/
       // Outputs
       .c                               (c_int[4:0]),            // Templated
       // Inputs
       .a                               (a_int[3:0]),            // Templated
       .b                               (b_int[3:0]));            // Templated

endmodule
```

wsnyder, 2021-11-11:
This is because it inserts a , for that last line, then cleans it up by editing after-the-fact to );.  I updated master to remove a space in the process. If tabs are used it'll still be wrong.

closed 2021-11-11 by wsnyder

fnJeff, 2023-03-28:
Finally had time to update to the latest version.  Apologies for the ridiculous delay.  Confirmed as fixed.  My eyeballs and OCD now love you.  Thanks!!!