veripool/verilog-mode #1753: Incorrect highlighting of words after "module" or "interface" in a comment

fnJeff, 2021-11-12:
Another highlighting issue I noticed.  It appears that if the word "module" of "iterface" appears in a comment, the word after it is highlighted using a particular face.  This seems to be the same face used after module/interface in non-comment context.  There are probably other related keywords that have this same effect, but these are the two I found hiding my existing code so far.

[image]

wsnyder, 2021-11-12:
verilog-font-lock-keywords-1 is what highlights this.  Normally the syntax table indicates in a comment so overrides this, not sure why it isn't.

gmlarumbe, 2022-06-19:
Hi,

This should be fixed in 883179f

According to https://github.com/Lindydancer/presentations/blob/master/font-lock-introduction.org:

> Font-lock keywords
> Flags
> 
> A HIGHLIGHT has two optional flag fields:
> 
> (SUBEXP FACE OVERRIDE LAXMATCH)
> OVERRIDE
> 
> OVERRIDE controls what happens when there already was a face at the location. If not present, the match is not applied. In most cases this is the desired effect, as we don’t want to mess up comments or strings. Legal values are “t”, “keep”, “append”, and “prepend”.
> 
> Hint: Prefer “prepend” over “t”, as it behaves better when the new and existing face provides different attributes, for example when the existing face specifies a background color and the new doesn’t.

For some reason it was set to `prepend` and was overriding comments and strings.

Before:

[image]

After:

[image]

gmlarumbe, 2022-06-19:
@fnJeff can you confirm it works for you as well?

gmlarumbe, 2022-07-09:
@wsnyder , does it work for you? 

I would probably make a PR but it contains very few changes and commit doesn't affect any test.

wsnyder, 2022-07-09:
If you want me to merge it, I'll need a pull request, thanks.

gmlarumbe, 2022-07-09:
I'm sorry, I got a bit confused wrt this issue.

I already pushed the changes into master in 883179f (3 weeks ago). I just wanted to make sure that it works fine also under XEmacs before closing the issue. 

You can try with previous snippet:
```verilog
// Face: `font-lock-function-name-face'
// module foo
// primitive foo
// class foo
// program foo
// interface foo
// package foo
// task foo
//
// Face: `font-lock-constant-face'
// function real foo
// function integer foo
// function time foo
```

closed 2022-08-07 by gmlarumbe

fnJeff, 2022-08-08:
Apologies, I was buried at work followed by vacation followed by reburying
at work.

Hopefully you were able to verify this well enough without me.

Thank you for the fix!  I look forward to it whenever I get the time to
update to a newer version.

fnJeff

fnJeff, 2023-03-28:
Finally had time to update to the latest version.  Apologies for the ridiculous delay.  Confirmed as fixed.  Thank you for this fix!!!