veripool/verilog-mode #1756: How to use defined "parameter FOR_NUMS = 8;" in @for loop

thfirst, 2021-12-07:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (You may want to attach output from 'M-x verilog-submit-bug-report')

Can you please check your github name is set to your real name (click on your avatar icon in upper right, then "settings" then "Name")?

I hope to use the defined "parameter" as the loop constant in the RTL as following:

module Test (/*AUTOARGS*/);
parameter FOR_NUMS = 8;
@for i = 0 to FOR_NUMS  //it is illegal for now
input [31:0] P@i@ ;
@endfor

Is there any work-around solution ?

thfirst, 2021-12-07:
Fixed some typo:
module Test (/*AUTOARGS*/);
parameter FOR_NUMS = 8;
@for i = 0 to FOR_NUMS     //  it is illegal for now__
input [31:0] P@i@ ;
@endfor

Is there any work-around solution ?

closed 2021-12-07 by thfirst

wsnyder, 2021-12-07:
There's intentionally no "for" loops in verilog-mode.

wsnyder, 2021-12-07:
Although note you could use AUTOINSERTLISP to write your own for loop or otherwise do anything that generates code.

thfirst, 2021-12-08:
I see. Thanks a lot.