veripool/verilog-mode #1757: Feature request: multiple regexp patterns be indexed in instance name  for AUTO_TEMPLATE

thfirst, 2021-12-09:
The standard @ sign in a template by default returns the first number in a instance name. If there are multiple regexp patterns in a instance name as following: there are 2 patterns
/* InstModule AUTO_TEMPLATE "u_\(a.*\)_\(b.*\)" (
     .a(@"(substring vl-cell-name 2)"_in[]),
     .b(@"(substring vl-cell-name 2)"_out[])
 );*/

Could these pattern be indexed as @0, @1, ..,  or other format, then they could be used in the AUTO_TEMPLATE?
If not, is there any work-aroud solution?

Thanks!

wsnyder, 2021-12-09:
We can't add @1 etc as a feature as that would break the current use that inserts "@" followed by "1"..

You can use the vl-cell-name and a lisp template, see an example here: https://github.com/veripool/verilog-mode/blob/master/tests_ok/more_params.v

closed 2021-12-09 by wsnyder

thfirst, 2021-12-09:
It works.
Thanks a lot!