veripool/verilog-mode #1758: Is there a variable to control in which column a port name appears?

cswfb, 2021-12-28:
I searched through the customizable variables listed on https://veripool.org/verilog-mode/help/ but did not find one that would control the column in which a port name declaration appears. It would be desirable to be able to control the column where the port names appear, similar to what the verilog-auto-inst-column variable does for the net name part of AUTOINST created pin.

Below is an example of the problem I am trying to solve. I would like all of the port names (clk, din, reset_n, dout) to be aligned in the same column. However, ports like din and dout that have bus ranges defined by parameters end up getting pushed out while signals without bus ranges (clk and reset_n) are not.

```
module example #
(
  parameter int DATA_WIDTH = 32
)
(
  /*AUTOINPUT*/
  // Beginning of automatic inputs (from unused autoinst inputs)
  input logic           clk,                    // To i_leaf of example_leaf.v
  input logic [DATA_WIDTH-1:0] din,             // To i_leaf of example_leaf.v
  input logic           reset_n,                // To i_leaf of example_leaf.v
  // End of automatics
  /*AUTOOUTPUT*/
  // Beginning of automatic outputs (from unused autoinst outputs)
  output logic [DATA_WIDTH-1:0] dout            // From i_leaf of example_leaf.v
  // End of automatics
);

  example_leaf #
    (/*AUTOINSTPARAM*/
     // Parameters
     .DATA_WIDTH                        (DATA_WIDTH))
  i_leaf
    (/*AUTOINST*/
     // Outputs
     .dout                              (dout[DATA_WIDTH-1:0]),
     // Inputs
     .clk                               (clk),
     .reset_n                           (reset_n),
     .din                               (din[DATA_WIDTH-1:0]));

endmodule

module example_leaf #
(
  parameter int DATA_WIDTH = 32
)
(
  input  logic                  clk,
  input  logic                  reset_n,
  input  logic [DATA_WIDTH-1:0] din,
  output logic [DATA_WIDTH-1:0] dout
);

endmodule
```

Ideally, the code would look something like this after expanding the Verilog AUTOs.
```
module example #
(
  parameter int DATA_WIDTH = 32
)
(
  /*AUTOINPUT*/
  // Beginning of automatic inputs (from unused autoinst inputs)
  input logic                   clk,            // To i_leaf of example_leaf.v
  input logic [DATA_WIDTH-1:0]  din,            // To i_leaf of example_leaf.v
  input logic                   reset_n,        // To i_leaf of example_leaf.v
  // End of automatics
  /*AUTOOUTPUT*/
  // Beginning of automatic outputs (from unused autoinst outputs)
  output logic [DATA_WIDTH-1:0] dout            // From i_leaf of example_leaf.v
  // End of automatics
);

  example_leaf #
    (/*AUTOINSTPARAM*/
     // Parameters
     .DATA_WIDTH                        (DATA_WIDTH))
  i_leaf
    (/*AUTOINST*/
     // Outputs
     .dout                              (dout[DATA_WIDTH-1:0]),
     // Inputs
     .clk                               (clk),
     .reset_n                           (reset_n),
     .din                               (din[DATA_WIDTH-1:0]));

endmodule
```

wsnyder, 2021-12-29:
I think this is controlled by this line

    (indent-to (max 24 (+ indent-pt 16)))

so there's no variable at present.  Perhaps you would be able to contribute a pull request to make this programmable, along with an associated test.

gmlarumbe, 2022-07-02:
As a workaround you could run `verilog-pretty-declarations` on any of the lines between the parenthesis of the port list.

afuller525, 2023-04-21:
Is this still an open item? I had the same question and stumbled upon this old ticket and was wondering if it might have been added to subsequent releases.

wsnyder, 2023-04-21:
This has not been changed, still open.