veripool/verilog-mode #1759: Symbol's value as variable is void

futurehome, 2022-01-07:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (You may want to attach output from 'M-x verilog-submit-bug-report')

I try to use the following template to convert part of the port name to lowercase, it gives the error: "Symbol's value as variable is void", the \1 has been replaced by the matched part in the port name, but seems it's been treated as variable, not string. Why?

.DFT_\(.*\) (@"(downcase \1)",

Can you please check your github name is set to your real name (click on your avatar icon in upper right, then "settings" then "Name")?

wsnyder, 2022-01-07:
There's an example on the FAQ: https://github.com/veripool/verilog-mode/blob/master/FAQ.rst#how-do-i-use-auto-template-to-lower-case-all-signals

But to answer your question you need to quote and escape the  \1

              .DFT_(.*) (@"(downcase \\"\1\\")"[]),

closed 2022-01-07 by wsnyder

futurehome, 2022-01-07:
Great thanks.