veripool/verilog-mode #1760: Incorrect indentation

sjalloq, 2022-01-21:
Hi,

I'm using verilog-mode with Xemacs in batch mode to run auto's and indentation.  The following testcase highlights two issues with the indentation.

1. The `output` keyword is split with a space
2. The top module instance indentation is broken within a for loop.

```systemverilog
module test (
input logic a,
output logic b
);

for (genvar i=0; i<2; i++) begin : g_slice

submod
u_sub_0 #(
.RESET_POL (RESET_POL)
)(
.clk     (clk),
.reset_n (reset_n),
.d       (d[i]),
.q       (q_0[i])
);

submod
u_sub_1 #(
.RESET_POL (RESET_POL)
)(
.clk     (clk),
.reset_n (reset_n),
.d       (q_0[i]),
.q       (q[i])
);

end

endmodule
```

Output after running verilog-batch-indent:

```systemverilog
module test (
             input logic a,
             o utput logic b
             );

  for (genvar i=0; i<2; i++) begin : g_slice

    submod
                u_sub_0 #(
                          .RESET_POL (RESET_POL)
                          )(
                            .clk     (clk),
                            .reset_n (reset_n),
                            .d       (d[i]),
                            .q       (q_0[i])
                            );

    submod
      u_sub_1 #(
                .RESET_POL (RESET_POL)
                )(
                  .clk     (clk),
                  .reset_n (reset_n),
                  .d       (q_0[i]),
                  .q       (q[i])
                  );

  end

endmodule
```

wsnyder, 2022-01-22:
This is not at all what I get, I'm using verilog-mode-version "2022-01-02-e326b90-vpo" and xemacs 21.4 patch 24.

However, I would caution you that xemacs hasn't been upgrading for over 20 years, so you should not be using it, move to Gnu Emacs.

Closing as assuming upgrading will fix it, if not feel free to comment/reopen.

sjalloq, 2022-01-26:
Sorry, I said Xemacs when I mean Emacs as I had just compiled the latest version of verilog-mode and had it in my mind.

I'm using GNU Emacs v27.2.  I've attached a full testcase for you.  You'll just have to edit the path to emacs in `util/verilog_indent`.  Run the example using `util/verilog_indent test.sv`.

[testcase.tgz.zip](https://github.com/veripool/verilog-mode/files/7942536/testcase.tgz.zip)

wsnyder, 2022-01-27:
I agree this is a problem. The workaround for now is to use "(setq verilog-auto-lineup nil)"

gmlarumbe, 2022-06-27:
Hi @sjalloq 

I was able to reproduce the issue in 535bdf3 with `indent-tabs-mode` set to nil. It seems it is fixed in df69ad1. Can you test it in your environment to see if everything works fine before merging into master?

Thanks!

closed 2022-07-11 by gmlarumbe