veripool/verilog-mode #1761: Computing AUTOs with ifdef'd parameter list fails.

sjalloq, 2022-02-18:
I've just been setting up a flow to run all our code through `verilog-batch-auto-diff` and have found some syntax that verilog-mode doesn't like.  The following testcase shows the code:

```verilog
module submod (/*AUTOARG*/);

    parameter TEST = 0;

    input logic clk;

endmodule

module test(/*AUTOARG*/);

	input logic clk;

    submod

    `ifdef PARAMS
    #( .TEST (1) )
    `endif

    u_submod (/*AUTOINST*/);

endmodule
```

And the output from running 

```
Processing test.sv
%Error: test.sv:20: Can’t locate ‘`endif’ module definition (Expanded macro to endif)
    Check the verilog-library-directories variable.
    I looked in (if not listed, doesn’t exist):
	test.sv
```

The code in question is pretty horrific so we will likely rewrite it anyway but wanted to report it.

wsnyder, 2022-02-18:
Verilog-mode intentionally allows "submod `mydefine" and `ifdef is equivalent to a define as far as this goes as Verilog-mode otherwise ignores ifdefs.  So this is unlikely to ever get fixed, but thanks for reporting as it might help others out.

I suspect putting the ifdef inside the #( will fix it.

closed 2022-02-18 by wsnyder

sjalloq, 2022-02-18:
OK, understood and thanks for the fast response.