veripool/verilog-mode #1762: AUTO using an ifdef that chooses between two module instances.

knofr, 2022-03-02:
I carefully read **FAQ** part, and know that _ifdef_ s directive ignored. But i see in **FAQ**, that similar example has been [covered](https://github.com/veripool/verilog-mode/blob/master/FAQ.rst#how-do-i-ifdef-select-between-modules). 
```verilog
module b1
  (
   input            clk,
   input            rst,
   input [3:0]      data_in,
   output reg [3:0] data_out );

  always@(posedge clk)
    if(rst)
      data_out <= 0;
    else
      data_out <= data_in;
endmodule // b1

module b2
  (
   input            clk,
   input            rst,
   input [3:0]      data_in,
   input            enable,
   output reg [3:0] data_out );

  always@(posedge clk)
    if(rst)
      data_out <= 0;
    else if(enable)
      data_out <= data_in;

endmodule // b2

`define ENABLE
module buff_data_top();

  /*AUTOWIRE*/;
  /*AUTOREGINPUT*/;
  
`ifdef ENABLE
 `define BUFFER b2
`else
 `define BUFFER b1
`endif

  `BUFFER buffer_inst(/*AUTOINST*/);

  
endmodule // buff_data_top
// Local Variables:
// eval:(verilog-read-defines)
// End:
```

After evaluating AUTO, buffer_inst is always equal to the module instance b1, regardless of the *ifdef* value.

```verilog
`define ENABLE
module buff_data_top();

  /*AUTOWIRE*/;
  // Beginning of automatic wires (for undeclared instantiated-module outputs)
  wire [3:0]            data_out;               // From buffer_inst of `BUFFER.v
  // End of automatics
  /*AUTOREGINPUT*/;
  // Beginning of automatic reg inputs (for undeclared instantiated-module inputs)
  reg                   clk;                    // To buffer_inst of `BUFFER.v
  reg [3:0]             data_in;                // To buffer_inst of `BUFFER.v
  reg                   rst;                    // To buffer_inst of `BUFFER.v
  // End of automatics
  
`ifdef ENABLE
 `define BUFFER b2
`else
 `define BUFFER b1
`endif

  `BUFFER buffer_inst(/*AUTOINST*/
                      // Outputs
                      .data_out         (data_out[3:0]),
                      // Inputs
                      .clk              (clk),
                      .rst              (rst),
                      .data_in          (data_in[3:0]));

  
endmodule // buff_data_top
// Local Variables:
// eval:(verilog-read-defines)
// End:
```
Using verilog-mode version 2021-10-14. Config :
```lisp
  (setq verilog-indent-level             2
        verilog-indent-level-behavioral  2
        verilog-indent-level-declaration 2
        verilog-indent-level-module	2
        verilog-auto-inst-vector         nil)
```

wsnyder, 2022-03-02:
Verilog-mode ignores ifdefs, but it doesn't ignore defines, which it processes from top of the file to the bottom.  The last define value was the b1 so that is what it uses when looking for the module name.

closed 2022-03-02 by wsnyder

knofr, 2022-03-02:
I understand that this is happening. But is there a possibility that the **AUTOINST** worked correctly and there was a module, depending on the 
```verilog 
`define ENABLE
```
?

wsnyder, 2022-03-02:
No, sorry.