veripool/verilog-mode #1772: verilog-mode-map rebinds delete and backspace keys

bjourne, 2022-06-12:
I don't think the mode should do that since many people (me included) already have custom keybindings defined for those keys. Afaict, most other major modes leave those keys alone.

wsnyder, 2022-06-13:
That code predates the git repository.  I can't explain it.  I suspect this should be removed but first can you try removing it and using it for a while to see if anything strange pops up?

bjourne, 2022-06-13:
Will do. I have some .emacs config to clear those keybindings for me.

MichaelTYMcNamara, 2022-09-03:
I would offer that it is my view that it save to remove these bindings.  Please check first, but my recollection is they were convenience things, importing ideas from XEmacs that were not in emacs (or vice versa)  Xemacs seems to be frozen for more than a decade now...