veripool/verilog-mode #1777: 'verilog-mark-defun' does not mark functions

gmlarumbe, 2022-06-29:
Hi,

According to the docstring of `verilog-mode` and `verilog-mark-defun` this function should mark functions. However, since function/task regexps do not belong to `verilog-defun-re` and `verilog-end-defun-re`, `verilog-mark-defun` marks whole modules, classes, interfaces, packages...

I believe that simply adding function/task regexps to `verilog-defun-re` would not easily work as this variable is used for indentation, completion, autos...

I have made some changes in 8d2aa7c that mark functions/tasks and do not break any test. However still looks a bit confusing to have to separate between `verilog-end-of-defun` and `(verilog-re-search-forward verilog-end-defun-re)` in different parts of the code to keep compatibility.

What would be the best approach here? Is this a feature that should be merged into master?

Thanks!

wsnyder, 2022-06-29:
I don't use this feature myself, but I think the behavior would make sense to jump to the function, then if done again jump to the container of the function (module/package etc) which should happen automatically as that's the next most upper thing.

As an alternative you could just try both regexps (functions and non) and see which point is found earlier and jump to that.

wsnyder, 2022-06-29:
Also can you please convert this to a pull request? Thanks

PR #1805 (2022-09-15): Fix behavior of verilog-mark-defun
Hi,

This PR fixes #1777

Implements the following behavior, as suggested by @wsnyder:

> I don't use this feature myself, but I think the behavior would make sense to jump to the function, then if done again jump to the container of the function (module/package etc) which should happen automatically as that's the next most upper thing.

It also enables the use of `narrow-to-defun`.

Thanks!
closed 2022-09-15 by gmlarumbe