veripool/verilog-mode #1794: AUTO_TEMPLATE : Convert the selected part of the regular expression to lowercase

azbycxvictor, 2022-07-30:
i want verilog-mode to only connect to output signals, not input port names that match the port regular expression, then use a Lisp format template to match and modify it:
for example: AWVLD_test is output port, the result my expected is 
.AWVLD_test  (awvld_m0_sig);

.\([A-Z]*\)_test (@"(if (equal vl-dir \\"output\\") \\"\1_m0_sig\\" \\"\\")");
the reuslt is:
.AWVLD_test  (AWVLD_m0_sig);

if i change the Lisp, it can't be compiled.
.\([A-Z]*\)_test (@"(if (equal vl-dir \\"output\\") \\"(downcase \1)_m0_sig\\"  \\"\\")");

wsnyder, 2022-07-30:
You need to quotes around strings only, not functions (e.g. downcase).  Try

             .([A-Z]*)_test (@"(if (equal vl-dir \"output\") (downcase \"\1_m0_sig\"))");

closed 2022-07-30 by wsnyder

azbycxvictor, 2022-08-01:
Thanks a lot! That worked.