veripool/verilog-mode #1796: AUTOINST not working with user defined port types of a sub-module.

rutshub, 2022-08-04:
Hi,
  I am not able to get proper instantiation using AUTOINST for submodule with user defined type ports. I have a sub module which has ports defined as user defined type structure. But when I expand AUTOINST I see that instantiation includes port type in the connectivity.
Below is some useful info. I want to check if verilog-mode.el version I am using is suppose to work for user defined type or I need to move to latest version of verilog-mode?

Emacs  : GNU Emacs 26.3 (build 1, x86_64-suse-linux-gnu, GTK+ Version 2.24.20)
 of 2019-09-19
Package: verilog-mode v2017-08-07-c085e50-vpo-GNU

current state:
==============
(setq
 verilog-active-low-regexp nil
 verilog-after-save-font-hook nil
 verilog-align-ifelse nil
 verilog-assignment-delay ""
 verilog-auto-arg-sort nil
 verilog-auto-declare-nettype nil
 verilog-auto-delete-trailing-whitespace nil
 verilog-auto-endcomments t
 verilog-auto-hook nil
 verilog-auto-ignore-concat nil
 verilog-auto-indent-on-newline t
 verilog-auto-inout-ignore-regexp nil
 verilog-auto-input-ignore-regexp nil
 verilog-auto-inst-column 40
 verilog-auto-inst-dot-name nil
 verilog-auto-inst-interfaced-ports nil
 verilog-auto-inst-param-value nil
 verilog-auto-inst-sort nil
 verilog-auto-inst-template-numbers nil
 verilog-auto-inst-vector t
 verilog-auto-lineup 'declarations
 verilog-auto-newline t
 verilog-auto-output-ignore-regexp nil
 verilog-auto-read-includes nil
 verilog-auto-reset-blocking-in-non t
 verilog-auto-reset-widths t
 verilog-auto-save-policy nil
 verilog-auto-sense-defines-constant nil
 verilog-auto-sense-include-inputs nil
 verilog-auto-star-expand t
 verilog-auto-star-save nil
 verilog-auto-template-warn-unused nil
 verilog-auto-tieoff-declaration "wire"
 verilog-auto-tieoff-ignore-regexp nil
 verilog-auto-unused-ignore-regexp nil
 verilog-auto-wire-type "logic"
 verilog-before-auto-hook nil
 verilog-before-delete-auto-hook nil
 verilog-before-getopt-flags-hook nil
 verilog-before-save-font-hook nil
 verilog-cache-enabled t
 verilog-case-fold nil
 verilog-case-indent 2
 verilog-cexp-indent 2
 verilog-compiler "echo 'No verilog-compiler set, see \"M-x describe-variable verilog-compiler\"'"
 verilog-coverage "echo 'No verilog-coverage set, see \"M-x describe-variable verilog-coverage\"'"
 verilog-delete-auto-hook nil
 verilog-getopt-flags-hook nil
 verilog-highlight-grouping-keywords nil
 verilog-highlight-includes t
 verilog-highlight-modules nil
 verilog-highlight-p1800-keywords nil
 verilog-highlight-translate-off nil
 verilog-indent-begin-after-if t
 verilog-indent-declaration-macros nil
 verilog-indent-level 3
 verilog-indent-level-behavioral 3
 verilog-indent-level-declaration 3
 verilog-indent-level-directive 1
 verilog-indent-level-module 3
 verilog-indent-lists t
 verilog-library-directories '("." "../../include" "../../../subip/sip/sb2ucie_IOSF_SBC_EP/source/rtl/iosfsbc/endpoint")
 verilog-library-extensions '(".v" ".sv")
 verilog-library-files '("")
 verilog-library-flags '("")
 verilog-linter "echo 'No verilog-linter set, see \"M-x describe-variable verilog-linter\"'"
 verilog-minimum-comment-distance 10
 verilog-mode-hook 'verilog-set-compile-command
 verilog-mode-release-emacs t
 verilog-mode-version "2017-08-07-c085e50-vpo-GNU"
 verilog-preprocessor "vppreproc __FLAGS__ __FILE__"
 verilog-simulator "echo 'No verilog-simulator set, see \"M-x describe-variable verilog-simulator\"'"
 verilog-tab-always-indent t
 verilog-tab-to-comment nil
 verilog-typedef-regexp "tier1$"
 verilog-warn-fatal nil
 )

Below is what I have at end of my SV file.
// Local Variables:
// verilog-library-directories:("." "../../include" "../../../subip/sip/sb2ucie_IOSF_SBC_EP/source/rtl/iosfsbc/endpoint")
// verilog-library-files:("")
// eval:(verilog-read-defines)
// eval:(verilog-read-includes)
// verilog-case-fold:nil
// verilog-typedef-regexp: "_tier1$"
// End:

Thanks,
Ruturaj.

wsnyder, 2022-08-07:
I suspect your user defined type doesn't match the regexp you provided:

           // verilog-typedef-regexp: "_tier1$"

If that's not the problem, please attach a complete self contained example that shows the problem.

closed 2022-08-07 by wsnyder

rutshub, 2022-08-09:
Hi Wilson,
    Please see attached files which is an example where I am not able to get AUTOINST instantiation as expected.
Thanks,
Ruturaj.
[sverilog.zip](https://github.com/veripool/verilog-mode/files/9286876/sverilog.zip)

wsnyder, 2022-08-09:
You need the verilog-typedef-regexp in all modules that are declaring signals (or set it globally).