veripool/verilog-mode #1801: emacs expansion of a large auto-templated file fails with no hints on where the error - DEBUG METHODS

engrvns, 2022-08-15:
Hi verilog-mode team

Problem: 
I have a large RTL with extensive AUTO_TEMPLATE being used.
Sometime when there are multiple AUTO_TEMPLATED edited the emacs verilog-mode expansion fails with the error as -

Below's a transcript of what happens -

```
emacs --batch -nw  RTL.v  -f verilog-mode -f verilog-batch-auto
Processing RTL.v
%Error: End of file during parsing <===
```

This happens when you use the below construct in the AUTO_TEMPLATE
.port (@"(if (equal vl-dir \\"output\\") \\"\\" \\"'0\\")"),

Note: The construct is correct for matching parentheses and the same code works in other parts of the file..

In these cases 

1. Other than onion-peel method commenting and uncommenting changes, how can we get verilog-mode to print out more detailed message to help localize the error.

1. Are there currently any such switches that can help debug this?

Thanks for the wonderful contribution to this community.

EVS

engrvns, 2022-08-15:
This happens with the code -
```
/* inst_module AUTO_TEMPLATE (
...
.port (@"(if (equal vl-dir \\"output\\") \\"\\" \\"'0\\")"),
...
);
*/

```

The expanded code still has the unexpanded piece of code
	 .all_dcs_idle (@"(if (equal vl-dir \\"output\\") \\"\\" \\"'0\\")"), //FIXME-dbg

And the error while expansion is -
%Error: End of file during parsing <===

wsnyder, 2022-08-15:
The example works fine for me, please check you are using the latest verilog-mode.el and if you are, please send a complete self-contained example that shows the problem.

One way to debug is Verilog-Mode will have expanded all of the lines up until the one that is the problem, so for example if port2 is the problem you'll see "port1" as having its AUTOINST expanded, then the rest of the file won't have anything expanded.

closed 2022-09-15 by engrvns