veripool/verilog-mode #1808: verilog-read-define

egorman44, 2022-09-12:
Hi folks, I'm trying to choose between two different modules with the same port list using the `ifdef statement:

```
`ifdef FPGA
 `define CHILD_INST child_0
`else
 `define CHILD_INST child_0
`endif

```

I have two issues(I've attached a code snippet to reproduce it):
1. It looks like I couldn't control `ifdef execution by adding or deleting thestatement :

```
`define FPGA
`ifdef FPGA
 `define CHILD_INST child_0
`else
 `define CHILD_INST child_0
`endif
```

2. AUTO_TEMPLATE doesn't work with `ifdef statement.

Here is the code I'm trying to execute:

```
module child_0
  (
   input  axis0_tdata_i,
   input  axis0_tvalid_i,
   output axis1_tdata_o,
   output axis1_tvalid_o,
   input  axis1_tready_i
   );
endmodule // child_0

module child_1
  (
   input  axis1_tdata_i,
   input  axis1_tvalid_i,
   output axis1_tready_o,
   output axis2_tdata_o,
   output axis2_tvalid_o,
   input  axis2_tready_i
   );
endmodule // child_1

module test_0();

   // !!!!!!!!!!!!!!!!!!!!!
   // `ifdef execution doesn't depend on whether the next line commented out or not
   // !!!!!!!!!!!!!!!!!!!!!
   
`define FPGA
   
`ifdef FPGA
 `define CHILD_INST child_0
`else
 `define CHILD_INST child_0
`endif

   // !!!!!!!!!!!!!!!!!!!!!
   // AUTO_TEMPLATE doesn't work 
   // !!!!!!!!!!!!!!!!!!!!!
   
   /*child_0 AUTO_TEMPLATE
     (.axis1_tdata_o(qwerty));
   */

   `CHILD_INST inst(/*AUTOINST*/
		    // Outputs
		    .axis1_tdata_o	(axis1_tdata_o),
		    .axis1_tvalid_o	(axis1_tvalid_o),
		    // Inputs
		    .axis0_tdata_i	(axis0_tdata_i),
		    .axis0_tvalid_i	(axis0_tvalid_i),
		    .axis1_tready_i	(axis1_tready_i));

endmodule // test_0

// Local Variables:
// eval:(verilog-read-defines)
// End:

```

wsnyder, 2022-09-12:
Ifdefs need to be done by hand, see https://github.com/veripool/verilog-mode/blob/master/FAQ.rst#why-do-the-autos-ignore-my-ifdefs

closed 2022-09-12 by wsnyder

egorman44, 2023-03-02:
Thank you @wsnyder for your response!

I just edited my previous message(port lists were different for child0 and child1, also **ifdef** condition picked the same module **child0**)
I decided to leave a solution I have to substitute blocks with the same port lists and process AUTO_TEMPLATE. 
Moreover, I was able to handle the situation where I have multiple instances of blocks(**child0** or **child1** or **CHILD_INST**) with a single AUTO_TEMPLATE

```
module child_0
  (
   input  axis0_tdata_i,
   input  axis0_tvalid_i,
   output axis1_tdata_o,
   output axis1_tvalid_o,
   input  axis1_tready_i
   );
endmodule // child_0

module child_1
  (
   input  axis0_tdata_i,
   input  axis0_tvalid_i,
   output axis1_tdata_o,
   output axis1_tvalid_o,
   input  axis1_tready_i
   );
endmodule // child_1

module test_0();

   // !!!!!!!!!!!!!!!!!!!!!
   // `ifdef execution doesn't depend on whether the next line commented out or not
   // !!!!!!!!!!!!!!!!!!!!!

`ifdef FPGA
 `define CHILD_INST child_0
`else
 `define CHILD_INST child_1
`endif

   /*
    `CHILD_INST AUTO_TEMPLATE  // !!!! Here you need to add `CHILD_INST instead of module name !!!
    child_0 AUTO_TEMPLATE
    (
    .\(axis[01]_t.*\)_[io](\1_@),
    );
   */

   `CHILD_INST inst0(/*AUTOINST*/
                     // Outputs
                     .axis1_tdata_o     (axis1_tdata_0),         // Templated
                     .axis1_tvalid_o    (axis1_tvalid_0),        // Templated
                     // Inputs
                     .axis0_tdata_i     (axis0_tdata_0),         // Templated
                     .axis0_tvalid_i    (axis0_tvalid_0),        // Templated
                     .axis1_tready_i    (axis1_tready_0));       // Templated

   child_0 inst1(/*AUTOINST*/
                 // Outputs
                 .axis1_tdata_o         (axis1_tdata_1),         // Templated
                 .axis1_tvalid_o        (axis1_tvalid_1),        // Templated
                 // Inputs
                 .axis0_tdata_i         (axis0_tdata_1),         // Templated
                 .axis0_tvalid_i        (axis0_tvalid_1),        // Templated
                 .axis1_tready_i        (axis1_tready_1));       // Templated

endmodule // test_0
```

Then the next question is:
Is it legal to have multiple AUTO_TEMPLATES within the same template? Or am I just lucky enough that it works for me? 
I'm asking because I couldn't find any information about it.

wsnyder, 2023-03-02:
Having multiple templates like that is supported, and intentionally done for cases like you showed.