veripool/verilog-mode #1809: Focusing an internal wire to be also declared as an output port

engrvns, 2022-09-13:
Refer to code below 

```
```
module top (...)

wire A;

modA instA (
  .out(A)
  ...);

modB instB (
   in(A)
...);

endmodule
```
```

In the example above, I would like internal wire A to be promoted as output of module top as well.
I have thousands of such wires which needs to ports..

Is there a simple way that I can force a wire matching a regex also as a port.

Thanks

wsnyder, 2022-09-13:
Perhaps /*AUTOOUTPUTEVERY*/?  See https://veripool.org/verilog-mode/help/#verilog-auto-output-every

engrvns, 2022-09-15:
I tried the simple example on the link and it works.
/*AUTOOUTPUTEVERY*/

But in the real design it does not work. Are there any conditions for this...
How do I debug this. I have documented the experiments I have done below

Here are the checks I did -
1. The wires which I am trying to force as output ports are generated by /*AUTOWIRE*/
2. The design is compile clean
3. There are no Verilog-ignore directives that can override this.
4. My design is organized as -

module test (/*AUTOARG*/)
/*AUTOINPUT*/
/*AUTOOUTPUT*/
/*AUTOOUTPUTEVERY*/

/*AUTOWIRE*/

<code>

endmodule

wsnyder, 2022-09-15:
If you can make a little test case showing what you get I can help debug.

engrvns, 2022-09-20:
A deeper debug showed that the issue was else where. On fixing an unrelated issue AUTOOUTPUTEVERY works

closed 2022-09-20 by engrvns