veripool/verilog-mode #1811: AUTOOUTPUTEVERY and AUTOWIRE create duplicate definitions for a signal - Possible bug

engrvns, 2022-09-20:
I am using AUTOOUTPUTEVERY to promote a set of internal signals also as output ports.
In my RTL, I have this as -

top (/*AUTOARG*/);
/*AUTOUTPUT*
/*AUTOUTPUTEVERY("test")*/
/*AUTOWRITE*/

The vexpanded rtl is -

top (...);
/*AUTOUTPUTEVERY("test")*/
//Begin of automatics
output logic test; <- Notice this!!!
//end of automatics

/*AUTOWIRE*/
//Begin of automatics
logic test; <-- Notice this !!!
/?End of automatics

Notice test is defined as logic in two place leading to a compilation error.

This seems to a BUG ...

wsnyder, 2022-09-20:
AUTOWIRE should not be used/needed when using AUTOOUTPUTEVERY.

closed 2022-09-20 by wsnyder

engrvns, 2022-09-20:
Thanks Wilson,

In this case, how do users handle a scenario where only some internal AUTOWIRES needs to be a small set output ports and others need to be auto wires.

Using ignore-outputs is going to be messy 
Is there a way to avoid generating AUTOWIRES for the same regex used for AUTOOUTPUTEVERY

wsnyder, 2022-09-20:
I suspect you want AUTOOUTPUT instead.

engrvns, 2022-09-21:
But AUTOOUTPPUT does not promote a signal that's consumed internally as an output port.
AUTOOUTPUTEVERY("regex")  seems to be a perfect solution for this