veripool/verilog-mode #1812: AUTOOUTPUTEVERY and AUTOWIRE create duplicate definitions for a signal - Possible bug

engrvns, 2022-09-22:
Refer Last comment in - https://github.com/veripool/verilog-mode/issues/1811
The issue got closed and the comments could have been missed.

@wsnyder > I suspect you want AUTOOUTPUT instead.
@user>
But AUTOOUTPPUT does not promote a signal that's consumed internally as an output port and hardcoding them defeats the purpose.
AUTOOUTPUTEVERY("regex") seems to be a perfect solution for this except  duplicate definitions which seems to be a bug.

wsnyder, 2022-10-05:
The problem here is AUTOOUTPUTEVERY is expanded after AUTOWIRE (and some others) which it needs to be as it wants to output what gets declared by AUTOWIRE (only if it is used but not if it isn't).  But this bug wants the opposite order, that is do AUTOOUTPUTEVERY first, but then later do AUTOWIRE.

Obviously both orders can't be satisfied.  So at this time I don't see how to implement this case correctly without some major rework I'm not ready to undertake.

wsnyder, 2022-10-05:
Note the preexisting autooutputevery_wire.v shows this exact case.