veripool/verilog-mode #1814: Verilog-mode with LSP?

unfrostedpoptart, 2022-09-28:
Sorry - this is more a help request than an issue with verilog-mode although I think this is the best place for it to be seen by the desired audience.

A while back ( [https://github.com/veripool/verilog-mode/issues/1710](url) ), I asked about using LSP with Verilog-mode but Wilson wasn't interested in adding support.  Now that I understand LSP a tiny bit better, I think I can ask a more intelligent question:

Is anyone using verilog-mode and LSP together and have a working configuration? From my slightly better understanding of LSP, verilog-mode would still be used for the wonderful AUTO functionality. It might also be in charge of indentation but I'm not positive on this. However, the font-lock stuff would all be disabled and LSP would handle this.  And, LSP would also deal with all the completion and links to definitions. Therefore, I don't think verilog-mode needs and changes to work with LSP - just a few setq-s.

So, I'm hoping someone here has set this up and can give installation and configuration tips!

gmlarumbe, 2023-01-29:
Hi @unfrostedpoptart 

You could take a look at `verilog-ext` package. It provides functions that ease the setup of `lsp` and `eglot` for `verilog-mode`:
- https://github.com/gmlarumbe/verilog-ext

unfrostedpoptart, 2023-01-29:
Thanks!  It's still about a hundred steps from the docs but I think I can follow it.